raspberrypi/documentation #2825: Deprecation warnings on building the documentation site

aallan, 2023-03-02:
After the PRs for dependency updates that @mudge merged the other day I’m now getting deprecation warnings on building the documentation site.

````
Deprecation Warning: Using / for division outside of calc() is deprecated and will be removed in Dart Sass 2.0.0.

Recommendation: math.div($spacing-unit, 2) or calc($spacing-unit / 2)

More info and automated migrator: https://sass-lang.com/d/slash-div

   ╷
40 │   margin-bottom: $spacing-unit / 2;
   │                  ^^^^^^^^^^^^^^^^^
   ╵
    ../../../../minima-2.5.1/_sass/minima/_base.scss 40:18           @import
    minima.scss 48:3                                                 @import
    /Users/aa/Repos/documentation/build/jekyll/assets/main.scss 1:9  root stylesheet
````

mudge, 2023-03-02:
See https://github.com/jekyll/minima/issues/709

aallan, 2023-03-02:
Thanks for the pointer @mudge

nelliemckesson, 2023-03-14:
@aallan given the contents of that thread Paul linked, would you be ok with just hiding those warnings until minima ships their next release?

aallan, 2023-03-14:
> @aallan given the contents of that thread Paul linked, would you be ok with just hiding those warnings until minima ships their next release?

Seems reasonable! 🤣

nelliemckesson, 2023-03-17:
This is temporarily handled by #2866 , but let's keep this issue open as a reminder to keep an eye on minima releases.

lurch, 2023-05-21:
According to the issue @mudge linked to above, this hasn't been fixed in upstream yet.