raspberrypi/documentation #2828: Out of sequence warnings

aallan, 2023-03-02:
On build we have the following out of sequence warnings,

````
asciidoctor: WARNING: functions.adoc: line 19: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: functions_vars.adoc: line 19: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: globals.adoc: line 19: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: globals_a.adoc: line 19: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: globals_b.adoc: line 19: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: globals_c.adoc: line 19: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: globals_d.adoc: line 19: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: globals_defs.adoc: line 19: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: globals_e.adoc: line 19: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: globals_eval.adoc: line 19: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: globals_f.adoc: line 19: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: globals_func.adoc: line 19: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: globals_func_a.adoc: line 19: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: globals_func_b.adoc: line 19: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: globals_func_c.adoc: line 19: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: globals_func_d.adoc: line 19: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: globals_func_e.adoc: line 19: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: globals_func_f.adoc: line 19: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: globals_func_g.adoc: line 19: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: globals_func_h.adoc: line 19: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: globals_func_i.adoc: line 19: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: globals_func_l.adoc: line 19: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: globals_func_m.adoc: line 19: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: globals_func_n.adoc: line 19: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: globals_func_p.adoc: line 19: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: globals_func_q.adoc: line 19: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: globals_func_r.adoc: line 19: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: globals_func_s.adoc: line 19: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: globals_func_t.adoc: line 19: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: globals_func_u.adoc: line 19: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: globals_func_v.adoc: line 19: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: globals_func_w.adoc: line 19: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: globals_func_x.adoc: line 19: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: globals_g.adoc: line 19: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: globals_h.adoc: line 19: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: globals_i.adoc: line 19: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: globals_l.adoc: line 19: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: globals_m.adoc: line 19: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: globals_n.adoc: line 19: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: globals_p.adoc: line 19: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: globals_q.adoc: line 19: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: globals_r.adoc: line 19: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: globals_s.adoc: line 19: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: globals_t.adoc: line 19: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: globals_u.adoc: line 19: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: globals_v.adoc: line 19: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: globals_w.adoc: line 19: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: globals_x.adoc: line 19: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: group__alarm.adoc: line 9: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: group__async__context__freertos.adoc: line 9: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: group__async__context__poll.adoc: line 9: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: group__async__context__threadsafe__background.adoc: line 9: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: group__channel__config.adoc: line 9: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: group__critical__section.adoc: line 9: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: group__cyw43__driver.adoc: line 9: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: group__cyw43__ll.adoc: line 9: section title out of sequence: expected level 1, got level 3
asciidoctor: WARNING: group__interp__config.adoc: line 9: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: group__lock__core.adoc: line 9: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: group__multicore__fifo.adoc: line 9: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: group__multicore__lockout.adoc: line 9: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: group__mutex.adoc: line 9: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: group__pico__bstack__cyw43.adoc: line 9: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: group__pico__lwip__arch.adoc: line 9: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: group__pico__lwip__freertos.adoc: line 9: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: group__pico__lwip__nosys.adoc: line 9: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: group__pico__stdio__semihosting.adoc: line 9: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: group__pico__stdio__uart.adoc: line 9: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: group__pico__stdio__usb.adoc: line 9: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: group__pio__instructions.adoc: line 9: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: group__queue.adoc: line 9: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: group__repeating__timer.adoc: line 9: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: group__sem.adoc: line 9: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: group__sleep.adoc: line 9: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: group__sm__config.adoc: line 9: section title out of sequence: expected level 1, got level 2
[... cut ...] ted level 1, got level 2
asciidoctor: WARNING: group__util__datetime.adoc: line 9: section title out of sequence: expected level 1, got level 2
asciidoctor: WARNING: group__util__pheap.adoc: line 9: section title out of sequence: expected level 1, got level 2
````

which are coming from the Doxygen import. Not amazingly important, and potentially not solvable. Or potentially not worth the time to solve? Feel free to tag as `wontfix` and close if you feel it's not worth our while @nelliemckesson.

lurch, 2023-03-02:
**Part** of that will be because of the way that some modules are nested inside other modules, and the way that AsciiDoctor assumes that the header-levels are strictly hierarchical. For example:
[image]

[image]

[image]

In the Doxygen-HTML these are all the same header-level (since each of `pico_cyw43arch`, `cyw43_driver` and `cyw43_ll` are each a module in their own right), but because of the way they're nested, AsciiDoctor gives them gradually smaller and smaller header-sizes (and hence _some_ of the warnings you're seeing above).

We had similar problems in the Doxygen XML->AsciiDoc->PDF version of the API-documentation, where some of the "nested" modules weren't visible in the Table-Of-Contents / Document-Outline.

aallan, 2023-05-07:
Still valid.

PR #2963 (2023-10-17): Fixing adoc out-of-sequence warnings for pico-sdk
Fixes #2828 . @lurch give it a try and let me know if you see anything unexpected!
closed 2023-10-17 by aallan