raspberrypi/documentation #2829: Doxygen: inter-page navigation isn't quite working properly

lurch, 2023-03-02:
If I'm looking at https://www.raspberrypi.com/documentation/pico-sdk/networking.html and then in the LHS-menu I expand the "High Level APIs" section, if I then click on the "pico_multicore -> Functions" menu-item, it goes to https://www.raspberrypi.com/documentation/pico-sdk/high_level.html#functions-5 but it doesn't auto-scroll to the right part of the page. On this new page (now that I'm looking at `high_level.html`) the LHS-menu says that the "pico_multicore -> Functions" menu-item is https://www.raspberrypi.com/documentation/pico-sdk/high_level.html#functions30 and clicking on that _does_ auto-scroll to the right part of the (same) page.

lurch, 2023-03-03:
Hmmm, I've also just spotted that if I do "open in new tab" on that https://www.raspberrypi.com/documentation/pico-sdk/high_level.html#functions30 link above, it also doesn't scroll to the right part of the page! So then I noticed that on the **right**-hand-side of the page, the funny-symbol (see screenshot below) on the "pico_multicore -> Functions" header has the URL https://www.raspberrypi.com/documentation/pico-sdk/high_level.html#rpipf840d967c7b8d6fef316 and this is a URL which _does_ scroll to the right part of the page when I "open in new tab".
[image]

aallan, 2023-03-17:
Fixed by #2865

nelliemckesson, 2023-09-15:
See #3117

commit e3bd1b7e (2023-03-17): fixes #2829
PR #3117 (2023-09-16): Scroll to a specific element if # is in the window url
This script will scroll to a specific element as indicated in the window url, e.g. https://www.raspberrypi.com/documentation/pico-sdk/high_level.html#functions30. The reason this wasn't working before was due to the toc/nav library we use (tocify), which creates its own unique id markers for every header in the page and then links to those (e.g. "functions30") instead of the ids that we generate for each heading ("rpip1234..."). Since these ids don't exist until after tocify runs, it creates some order-of-operations issues when loading the [... cut]
closed 2023-09-16 by aallan