raspberrypi/documentation #2848: Doxygen: Curate which headings appear in the TOC

nelliemckesson, 2023-03-07:
From Graham:

> Always bugged me about the default Doxygen, but move the Detailed Description to the top… perhaps we want “Description”, “Members” or “Index” basically the list of data structures, typedefs functions etc - they can keep their headings… I don’t think having a section for the individual “Function Documentation” helps on the left, you are either going to search, or click from the “members"

Preliminary thoughts: *most* of the headings are based directly on what's coming from doxygen. In order to remove certain headings from the TOC, we need some predictable rules about which headings should be removed. This can be as simple as an exact string match, or more complex (e.g., headings that are tagged with a certain class in doxygen), but whatever the rule is, it needs to be fixed and consistent.

aallan, 2023-05-14:
Still pending.

aallan, 2023-07-30:
Still a thing.

aallan, 2023-10-01:
Still pending.

aallan, 2023-12-03:
Still (still) a thing.

nathan-contino, 2024-02-12:
Still needs doing.

nathan-contino, 2024-04-15:
Not so fast!