raspberrypi/documentation #2894: Updating a Codec Zero's firmware doesn't work

GuillaumeRossolini, 2023-04-15:
Hi,

Following my attempt at improving the documentation (see my initial https://github.com/raspberrypi/documentation/pull/2891), I learned that my audio hat wasn't properly set up at the firmware level and that the documentation doesn't need the changes I thought it did (see the subsequent https://github.com/raspberrypi/documentation/issues/2892).

So I was instructed to follow the "Updating your firmware" documentation here:
cf. https://www.raspberrypi.com/documentation/accessories/audio.html#updating-your-firmware

It appeared to work but also not work, and that's where I'm stuck.
* the `rpi-audio-flash` package showed me a successful message;
* however, neither the `cat /proc/device-tree/hat/vendor` command or the documented Codec Zero setup work after a reboot

As mentioned in the other thread, my hat is a pre-rebranding IQaudIO board from 2021 or earlier, rather than the Codec Zero shown in the documentation. As fas as I understand it, there are no electronic changes between the two, nothing more than the color and the written logo.

I followed the documentation and, since my hat already has the bridge gapped, I don't have to worry about write protection. The  `rpi-audio-flash` package was installed without any issue, and it ran also without any issue. I got a message on screen, saying that the firwmare was updated.

However, after a reboot, the instructions that were supposed to let me use the hat, didn't work. The steps I tried to suggest in https://github.com/raspberrypi/documentation/pull/2891 do work, though.

So this is where I'm at. I don't know if the instructions to update the firmware are incomplete, or if the firmware updating package falsely announced the successful completion of the firmare, or what.

I have a way to make my hat work fine, just with the old branding and with more complex setup than should be required (undocumented too, as shown by my rejected pull request mentioned above).

Looking for help. I'm happy to stay with my current setup, but I'm also happy to try to figure this out.

GuillaumeRossolini, 2023-04-17:
Hi and thanks for the update.

Like I said in the other issue, these settings don't work. The `dtoverlay` needed for my hat is referenced in the previous issue:
`dtoverlay=iqaudio-codec`

This also doesn't address how to flash the EEPROM so that the new documentation works. Or am I missing something?

GuillaumeRossolini, 2023-06-25:
if anyone would like to jump in that would be great

aallan, 2023-07-17:
Fixed buy #3014?

lurch, 2023-07-17:
Still pending on an additional tweak to https://github.com/raspberrypi/Pi-Codec - just waiting for @tobybroberts to be given write-access to that repo.

GuillaumeRossolini, 2023-08-05:
@aallan @lurch I'm not sure what should have happened, but what I got was an error from the rpi-audio-flash tool:
> Failed to detect an EEPROM device.

It's honest about it now ;)

Here's what I did on the same rpi+hat as before:
```
$ sudo apt update
[...]
$ sudo apt upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  avahi-daemon base-files bind9-host bind9-libs containerd grep isc-dhcp-client isc-dhcp-common libavahi-common-data
  libavahi-common3 libavahi-core7 libcamera-apps-lite libcamera0 libgstreamer-plugins-base1.0-0 libncurses6
  libncursesw6 libpam-systemd libssl1.1 libsystemd0 libtinfo5 libtinfo6 libudev1 libwebp6 libwebpdemux2 libwebpmux3
  libx11-6 libx11-data libxml2 libxpm4 ncurses-base ncurses-bin ncurses-term needrestart openssl python3-libcamera
  python3-picamera2 raspberrypi-sys-mods rpi-eeprom systemd systemd-sysv systemd-timesyncd traceroute tzdata udev
44 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
[...]
Unpacking rpi-eeprom (16.1-1) over (16.0-1)
[...]
$ sudo apt upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

$ sudo apt dist-upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

$ dpkg -l rpi-audio-utils
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name            Version      Architecture Description
+++-===============-============-============-=======================================
ii  rpi-audio-utils 0.1          all          Utilities for Raspberry Pi Audio boards
$ dpkg --verify rpi-audio-utils && echo ok || echo KO
ok
$ dpkg --audit rpi-audio-utils && echo ok || echo KO
ok
$ sudo reboot now
```

There is a log file after running *sudo rpi-audio-flash*:
> $ time sudo rpi-audio-flash && echo ok || echo KO
When reporting issues with this script, please attach /tmp/rpi-audio-WSYNK1qdPI.log
real    0m2.902s
user    0m0.286s
sys     0m0.344s
KO
$ cat /tmp/rpi-audio-WSYNK1qdPI.log
This will attempt to talk to an eeprom at i2c address 0x50. Make sure there is an eeprom at this address.
This script comes with ABSOLUTELY no warranty. Continue only if you know what you are doing.
Reading...
dd: failed to open '/sys/class/i2c-adapter/i2c-9/9-0050/eeprom': No such file or directory
Closing EEPROM Device.
Error doing I/O operation.

Which indeed does not exist
```
$ ls -alh /sys/class/i2c-adapter/i2c-9/9-0050
ls: cannot access '/sys/class/i2c-adapter/i2c-9/9-0050': No such file or directory
$ ls -alh /sys/class/i2c-adapter/i2c-9
lrwxrwxrwx 1 root root 0 Aug  5 11:59 /sys/class/i2c-adapter/i2c-9 -> ../../devices/platform/9.i2c/i2c-9
$ ls -alh /sys/devices/platform/9.i2c/i2c-9
total 0
drwxr-xr-x 4 root root    0 Aug  5 11:59 .
drwxr-xr-x 4 root root    0 Aug  5 11:59 ..
--w------- 1 root root 4.0K Aug  5 11:59 delete_device
lrwxrwxrwx 1 root root    0 Aug  5 12:04 device -> ../../9.i2c
drwxr-xr-x 3 root root    0 Aug  5 11:59 i2c-dev
-r--r--r-- 1 root root 4.0K Aug  5 11:59 name
--w------- 1 root root 4.0K Aug  5 11:59 new_device
lrwxrwxrwx 1 root root    0 Aug  5 12:04 of_node -> ../../../../firmware/devicetree/base/i2c@9
drwxr-xr-x 2 root root    0 Aug  5 12:04 power
lrwxrwxrwx 1 root root    0 Aug  5 11:59 subsystem -> ../../../../bus/i2c
-rw-r--r-- 1 root root 4.0K Aug  5 11:59 uevent
```

lurch, 2023-08-08:
ping @pelwell , as he's much more familiar with the low-level details of this than I am.

pelwell, 2023-08-08:
Holding the Codec Zero so that the labels are the right way up, find the words "Raspberry Pi (C) 2021". If you look to the left you'll see the label "WP", and immediately to the left of this there are two pads (silver squares). Is there a small rectangular component linking these pads together? If not, you'll need to find some way to short them together while running the script.

GuillaumeRossolini, 2023-08-08:
Ok so since I was just trying to improve the documentation for other people who might have the same board that I have, theirs might not be faulty and I can stay at I am now.
Thanks for the help!

closed 2023-08-08 by GuillaumeRossolini