raspberrypi/documentation #2895: openssl passwd -6 command not work on macOS

xgqfrms, 2023-04-16:
# `openssl passwd -6` command does not work on macOS 

```sh
# Linux
$ openssl passwd -6

```

https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/computers/configuration/headless.adoc#configuring-a-user

aallan, 2023-04-16:
I'm not sure why you're asking a macOS question on the Raspberry Pi documentation site?

xgqfrms, 2023-04-16:
In my opinion it's just that the documentation doesn't describe it clearly.

cmitu, 2023-04-20:
@xgqfrms but you're not actually using `openssl`, but a fork - LibreSSL.

The latest version of `openssl` is 3.1.0, which doesn't have the issue you're describing:
```
$ openssl version
OpenSSL 3.1.0 14 Mar 2023 (Library: OpenSSL 3.1.0 14 Mar 2023
$ echo test | openssl passwd -6 -in -
$6$8L2unG1nB1wI46pM$vk2WctBAKSZNYunjjn.P4PjHEj62CN.SOZW8gAkYHpwRsP.wvWuy1Lytrpo1PPi39Ryha51fm62JyFmv5wi0Q0
```

closed 2023-04-20 by xgqfrms