raspberrypi/documentation #2903: Question over config for use with Pi3 in Docs

grbd, 2023-05-01:
I've recently been trying to do a custom aarch64 kernel build for an rpi3
Something I noticed within the docs - https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/computers/linux_kernel/building.adoc

It mentions
`For Raspberry Pi 3, 3+, 4, 400 and Zero 2 W, and Raspberry Pi Compute Modules 3, 3+ and 4 default 64-bit build configuration`
```
cd linux
KERNEL=kernel8
make bcm2711_defconfig
```

Looking at `arch/arm64/configs`
I'm just wondering if for the Pi 3 and below if it should instead be `bcmrpi3_defconfig`
And for the Pi4 be `bcm2711_defconfig`

Does this sound correct?

lurch, 2023-05-02:
ping @pelwell

pelwell, 2023-05-02:
The instructions are correct. `bcmrpi3_defconfig` started out as a user-contributed 64-bit configuration file back in the Pi 3 era. We try to keep it up-to-date when new modules are enabled, but we only ship one 64-bit kernel build and that uses `bcm2711_defconfig`.

closed 2023-05-02 by lurch

grbd, 2023-05-02:
Ok thanks for the clarification