raspberrypi/documentation #2904: VNC changes

aallan, 2023-05-02:
Presumably due to the switch from X11 to Wayland, VNC has changed from RealVNC to https://github.com/any1/wayvnc.

bjepson, 2023-05-02:
Seems like there is no longer a VNC icon in the task bar either.
[image]

aallan, 2023-05-02:
> Seems like there is no longer a VNC icon in the task bar either. 

@ghollingworth Is this a bug or a feature?

aallan, 2023-05-03:
This section will need extensive reworking. https://www.raspberrypi.com/documentation/computers/remote-access.html#virtual-network-computing-vnc

ghollingworth, 2023-05-04:
@spl237 I think VNC will be broken with Wayland?

Don't we now have to use wayvnc?

Gordon

ghollingworth, 2023-05-04:
Ah...  I should start by reading the contents of the original issue!!!

ghollingworth, 2023-05-04:
We should probably talk to RealVNC about whether they would be able to support the Wayland interface a-la wayvnc

spl237, 2023-05-04:
Yup, RealVNC doesn't work with wayland; wayvnc is included and should be what gets enabled if you turn on VNC in rc-gui.

I've already raised the wayland issue with RealVNC; they seemed pretty non-committal. I didn't get the impression it would be a high priority for them!

aallan, 2023-05-15:
From Simon via @lurch 

> Apparently Wayland has better "sandboxing" of applications, so each running application isn't aware of any other running applications, and this means that for any applications which try to display a "System Tray" icon (e.g. VNC or VLC), this system-tray icon will no longer appear.

I'm still concerned that this is a safeguarding issue given deployment in classrooms.

aallan, 2023-05-15:
From Simon via @lurch 

> RealVNC (which was used in Bullseye) doesn't work with Wayland, so you need to use wayvnc instead. And apparently this only works with "newer" VNC clients like TigerVNC (?)

Not yet tested on headless deployment.

spl237, 2023-05-15:
> From Simon via @lurch
>
>
> I'm still concerned that this is a safeguarding issue given deployment in classrooms.

As above - there is nothing we can do about it. The system tray functionality which was used to display the icon previously has been removed completely under Wayland and not replaced with anything else that can do the same thing. We cannot show an icon to indicate that VNC is active (not unless someone writes a dedicated plugin to offer VNC functionality, which is a rather large task...)

spl237, 2023-05-15:
> From Simon via @lurch
> 
> > RealVNC (which was used in Bullseye) doesn't work with Wayland, so you need to use wayvnc instead. And apparently this only works with "newer" VNC clients like TigerVNC (?)
> 
> Not yet tested on headless deployment.

Tested by me on headless this morning - seems to work fine for me with TigerVNC even with no HDMI connected. One small bug found which prevented panel from starting which has now been fixed and will be in the repo later on today.

JamesH65, 2023-05-15:
> As above - there is nothing we can do about it. The system tray functionality which was used to display the icon previously has been removed completely under Wayland and not replaced with anything [...]

Could a minimal plugin be written whose entire raison d'etre is to display icons on the taskbar, then other apps can talk to that plugin via D-Bus or whatever it's called to get their icons displayed?

spl237, 2023-05-15:
> Could a minimal plugin be written whose entire raison d'etre is to display icons on the taskbar, then other apps can talk to that plugin via D-Bus or whatever it's called to get their icons displayed?

That would, in other words, be a system tray!

There are mechanisms which exist for some interapp communication within Wayland, but nothing like this has been implemented as a part of Wayfire. It would need creating from scratch, and would be doing something which the likes of Gnome have decreed should not happen any more. 

Going forward, if an icon is wanted on the taskbar, then that icon is created by a dedicated plugin which we load as part of the taskbar config, and which probably then actually needs to be the application which is doing the thing in question.

lurch, 2023-05-15:
> RealVNC (which was used in Bullseye) doesn't work with Wayland, so you need to use wayvnc instead. And apparently this only works with "newer" VNC clients like TigerVNC (?)

@tobybroberts was doing some testing last week, and he said that the RealVNC client seemed to work fine with the wayvnc server.

tobybroberts, 2023-05-15:
Further to the above from @lurch  - If no HDMI cable is attached RealVNC client still works but you lose the task bar completely, it just appears as a black bar.

spl237, 2023-05-15:
Yes, found and fixed that problem this morning - will be in apt in a few hours. Sent from my iPhone

aallan, 2023-07-13:
The current installation of VNC in the Bookworm beta release doesn't implement user/pass by default. Need to check if this gets implemented, and whether encryption is implemented by the new server. See https://forums.raspberrypi.com/viewtopic.php?t=353627 for details.

ghollingworth, 2023-07-13:
There is a pam option for WayVNC, not sure if it works yet...

spl237, 2023-07-13:
Have you tried TigerVNC? That was about the only thing I could get to connect to it in my testing. https://tigervnc.org Not sure I tried with keys though.

aallan, 2023-07-13:
> Having trouble configuring encrypted communication. If you provide WayVNC with an RSA4096 key it becomes very secure indeed because I can't so far find anything that'll connect!

A secure computer is one that's turned off, and put in a hole, and the hole filled with concrete. Even then it's not that secure.

ghollingworth, 2023-07-13:
Finally got it working...  Basically you need to:

openssl req -x509 -newkey rsa:4096 -sha256 -days 3650 -nodes \
	-keyout key.pem -out cert.pem -subj /CN=localhost \
	-addext subjectAltName=DNS:localhost,DNS:localhost,IP:127.0.0.1

I changed localhost to raspberry (the hostname of the pi on my desk!) and 127.0.0.1 to the IP address.  Not sure whether this would still work if the IP address changed!

then I've got a config file (normally stored in .config/wayvnc/wayvnc.cfg

address=0.0.0.0
enable_auth=true
enable_pam=true
private_key_file=/home/gordon/.config/wayvnc/key.pem
certificate_file=/home/gordon/.config/wayvnc/cert.pem

Then using Tigervnc on Windows to connect to this...  It will ask you for the username and password correctly

spl237, 2023-07-27:
Finally got around to implementing this in raspi-config.

Now, when you enable VNC for Wayfire, it generates an SSL key and sets the config file accordingly.

With regards changing the IP address, I just configured for my Ethernet IP and then set up wifi on a different address and disconnected the Ethernet cable. It all still connects fine; the only change is that it warns you that the hostname does not match the server certificate and checks if you want to continue connecting. Same happens if you change the Pi hostname.

S.

bjepson, 2023-08-01:
Thanks, @spl237! I tried connecting from RealVNC Viewer on my Mac, but I get this error:

> Unable to connect to VNC Server using your chosen security setting. Either upgrade VNC Server to a more recent version from RealVNC, or select a weaker level of encryption.

I am able to connect with TigerVNC, though. But I'm wondering if we should consider an encryption method that's compatible with RealVNC. From https://www.realvnc.com/en/blog/a-guide-to-end-to-end-encryption-e2ee:

> When using VNC Connect, you can rest assured that your data is protected by encryption.  All connections are encrypted end-to-end using up to 256-bit AES, 2048-bit RSA keys and perfect forward secrecy. This means that sessions are entirely private to you, now and in the future.

Thanks,

Brian

spl237, 2023-08-01:
TigerVNC was the only client I could get to connect to wayvnc, with or without encryption. I couldn't get a connection from RealVNC with any settings I tried even before adding the encryption. I may, of course, have been doing something wrong...

bjepson, 2023-08-01:
Thanks, @spl237. Curiously enough, I had been connecting to WayVNC with RealVNC prior to this. But I wasn't doing anything unusual, at least not that I was aware of. Is there a way for me to change it so it uses 2048 bit RSA and see if that makes a difference?

ghollingworth, 2023-08-02:
Previously, you were using unencrypted communication….

Wayvnc only supports this method of comms…. I’ve got in touch with the developer though who may be interested in developing it for us.

bjepson, 2023-08-02:
Thanks @ghollingworth, I mentioned that RealVNC had worked without encryption because @spl237's experience was that it hadn't worked at all (even without encryption).

What I'm wondering is whether changing from 4096 bit RSA to 2048 bit RSA would make a difference, because the RealVNC website says it handles up to 2048 bit.

ghollingworth, 2023-08-02:
Unfortunately, Wayvnc doesn’t…. Which is why I’ve been in contact with the
developer

bjepson, 2023-08-02:
I see, thanks. So even if we generated the key with `rsa:2048`, wayvnc would still use an encryption method that's incompatible with RealVNC.

Thanks,

Brian

bjepson, 2023-08-02:
Yeah, [here's the closed issue that addresses this](https://github.com/any1/wayvnc/issues/101):

> Neat VNC and RealVNC just don't support the same authentication protocol extensions. RealVNC's extension has not been shared with the public, so we cannot implement it without going through the trouble of reverse engineering it. Open extensions that are supported by RealVNC are easily defeated and should not be implemented in Neat VNC as they would provide a false sense of security.
> 
> I recommend ssh tunnelling.

aallan, 2023-09-04:
@nathan-contino You'll need to figure out what's going on here. You'll need to check the Bookworm beta forum (ping @LizUpton if you don't have access to this already) as the state of the art for this is there.

aallan, 2023-09-12:
Last issue before the mother issue can be closed!

aallan, 2023-09-14:
@nathan-contino Can confirm RealVNC still doesn't work out of the box. Using TigerVNC on my Mac. 

Need to select "show dot when no cursor" before connecting, because it doesn't seem to pick up the mouse cursor when connected otherwise.

android-808, 2023-09-14:
@aallan Regarding your safeguarding concern above.

I understand this is more about the documentation, but I feel you raised a valid security concern here.  Although the VNC connection has to be manually enabled, without some kind of user notification they may not be aware that someone has managed to access their system.  For a personal system you would have manually enabled it but in a environment where the user is using a shared/public system (classroom, library, cafe), they wouldn't necessarily even be aware that it has been enabled.  As mentioned above, a dedicated plugin to track VNC status/connections would be the ideal solution here to provide details to the user but would take someone a considerable effort to achieve.

https://github.com/any1/wayvnc/blob/master/wayvncctl.scd has a simple shell script to get messages from wayvncctl.  As a stop gap, would running something like this in the background and generating a desktop notification on connection/disconnection, like RealVNC currently does, be a quick, sufficient remedy to the concern.

spl237, 2023-09-14:
We don't show any notification when SSH is enabled, or an SSH connection is in progress. That is even more of a potential security risk, as if someone is connected to a remote desktop, they cannot do anything without the user seeing it - moving the mouse, opening applications etc will all be seen on the local screen. Whereas with SSH, anything the remote user is doing is completely hidden.

Anyone using either SSH or VNC has to have manually enabled it. They ought to know that they have done so, and anyone remote connecting to the system still has to have a set of valid login credentials to be able to access it.

If a system is available to the public in order for someone to secretly locally enable VNC or SSH, then it is also vulnerable to any number of other security problems - someone could have installed a keylogger or other Trojan, for example, or installed some other form of remote access software that we don't even know about. And if someone is using such a publicly-accessible system for anything personal or sensitive, then they really need to learn that they shouldn't. Putting up an icon to show there is a VNC connection addresses a single one of the multiplicity of security concerns in this scenario, and doesn't therefore seem that useful. Not least because, as above, SSH is just as easy to enable and presents a far more serious potential threat vector.

ebenupton, 2023-09-14:
I agree with Simon. This isn't a material incremental security risk.

aallan, 2023-09-20:
So we'll need a NOTE block along these lines @nathan-contino,

NOTE: In previous versions of Raspberry Pi OS the RealVNC client software was used to connect to the VNC server on your Raspberry Pi. In the latest version of Raspberry Pi OS, the VNC server is now WayVNC, and TigerVNC is now the required client software.

Not exactly that, but sort of like that.

aallan, 2023-09-20:
Last remaining issue, and we can close out the Bookworm milestone!

closed 2023-09-22 by aallan

ridwanrizvi, 2023-10-26:
RealVNC seems to be back I just did a fresh flash last night.

spl237, 2023-10-26:
RealVNC is not likely to support Wayland in the near future - it certainly doesn't support it now! So yes, you are getting RealVNC because you are running X11 on a Pi 3.

PR #3127 (2023-09-22): Update VNC documentation to support Bookworm VNC stack
Closes #2904 

* Updated server-side instructions to start the WayVNC server using the new options in RPi Configuration
* Took the liberty of making the "find your IP address" documentation a little more straightforward for the average user (the previous version hid the easiest command in a NOTE!), and to add a new way of finding your IP address using `nmcli` that I discovered when I worked on the network configuration section
* Removed instructions for removed features:
  * RealVNC-specific cloud connections
  * experimental single-applic [... cut]