raspberrypi/documentation #2925: Tutorial: How to use a Raspberry Pi in kiosk mode

tobybroberts, 2023-05-11:
Tutorial does not run on 64-bit Raspberry Pi Bookworm - code currently relies heavily on X environment.

aallan, 2023-05-15:
Works totally differently under Wayland

Edit .config/wayfire.ini, and add another line in the [autostart] section. The name at the start of the line can be anything you want, followed by an equals sign, followed by the command you want to execute. So something like:

```
[autostart]
panel = wf-panel-pi
background = pcmanfm --desktop --profile LXDE-pi
xdg-autostart = lxsession-xdg-autostart
chromium = chromium-browser "[example.com/example/](http://example.com/example/)" --kiosk --noerrdialogs --disable-infobars --no-first-run --ozone-platform=wayland --enable-features=OverlayScrollbar
```

tobybroberts, 2023-05-24:
Tutorial rewritten using changes to config/wayfire.ini file and new bash script utilising wtype instead of xdotool for key strokes.

aallan, 2023-10-11:
Deployed

aallan, 2023-10-11:
deployed

michaelsage, 2023-10-16:
Hi, I have followed the tutorial and it kinda works. When the pi boots the browser is opened but not displayed on the screen. I have to unplug and replug the hdmi connection and then the browser full screens fine. It doesn't matter which port the hdmi is plugged into the issue is the same, it also doesn't matter which port you replug the hdmin into. This is a fresh install of bookworm, I've tried it on multiple monitors. Thanks

jeremyfuksa, 2023-10-19:
> Hi, I have followed the tutorial and it kinda works. When the pi boots the browser is opened but not displayed on the screen. I have to unplug and replug the hdmi connection and then the browser full screens fine. It doesn't matter which port the hdmi is plugged into the issue is the same, it also doesn't matter which port you replug the hdmi into. This is a fresh install of bookworm, I've tried it on multiple monitors. Thanks
> 
> EDIT: Weirdly after the latest updates this now appears to be working as expected. Will leave this comment here for others.

I followed the tutorial using a fresh Bookworm install and I still need to do the HDMI re-plug fix to see the browser when the Pi gets to its desktop.

michaelsage, 2023-10-24:
So mine appears to start properly now (and still does), however Chromium crashes quite a lot and doesn't restart automatically so needs a reboot, I would say this happens 2 or 3 times a day

mrhighstone, 2023-10-31:
Same issue here on a Pi4 with latest Bookworm release, including updates. Need to replug the HDMI cable and then the browser is shown in Kiosk mode. Otherwise, the browser is visible on the menu bar but clicking on it doesn't show the browser.

Two differences noted:
1) I installed the 64bit edition, is there a reason why the 32bit is explicitly mentioned?
2) The default wayfire.ini on this fresh Bookworm install does not have the mentioned initial [autostart] section, but is limited to the sections command, input-device:xyz, input and idle