raspberrypi/documentation #2928: Tutorial: How to build a Raspberry Pi cluster

aallan, 2023-05-15:
The [cluster tutorial](https://www.raspberrypi.com/tutorials/cluster-raspberry-pi-tutorial/) makes extensive use of networking, all of which is going to change hugely due to the switch to Network Manger.

aallan, 2023-07-16:
Still needs doing.

aallan, 2023-09-17:
WIll probably be a post-Bookworm release task. But it still needs doing.

BrookeDot, 2024-01-30:
Hi @aallan is there a TL;DR: version on what the switch to Network Manager means for this tutorial and clustering in general on the Pi 4/5?

vlzambr2, 2024-08-11:
Commenting on this just to make sure it doesn't go stale again 🙈 . Definitely in need of this tutorial, my cluster is going to be stuck on Bullseye until a beginners' guide to clustering with Network Manager is published.

aallan, 2024-08-12:
Sorry. Forgot to update the issue. The tutorial has been already been updated for Bookworm.

closed 2024-08-12 by JamesH65