raspberrypi/documentation #2937: Hardware: check that our Audio HATs instructions work on Bookworm

lurch, 2023-05-16:
https://www.raspberrypi.com/documentation/accessories/audio.html

lurch, 2023-07-18:
Tested the instructions with a Raspberry Pi DAC+ - all works fine on Bookworm.

lurch, 2023-08-09:
Reopening. Just tested the [Codec Zero](https://www.raspberrypi.com/documentation/accessories/audio.html#extra-codec-zero-configuration) part of the instructions, and the `sudo alsactl restore -f /home/pi/Pi-Codec/Codec_Zero_OnboardMIC_record_and_SPK_playback.state` part fails with:
```
alsa-lib main.c:1541:(snd_use_case_mgr_open) error: failed to import hw:0 use case configuration -2
No state is present for card Headphones
alsa-lib main.c:1541:(snd_use_case_mgr_open) error: failed to import hw:0 use case configuration -2
/usr/share/alsa/init/bcm2835:20: control element not found
No state is present for card Headphones
alsa-lib main.c:1541:(snd_use_case_mgr_open) error: failed to import hw:1 use case configuration -2
No state is present for card vc4hdmi
alsa-lib main.c:1541:(snd_use_case_mgr_open) error: failed to import hw:1 use case configuration -2
Found hardware: "vc4-hdmi" "" "" "" ""
Hardware is initialized using a generic method
No state is present for card vc4hdmi
alsa-lib main.c:1541:(snd_use_case_mgr_open) error: failed to import hw:2 use case configuration -2
```

lurch, 2023-09-04:
Issue in bookworm-feedback https://github.com/raspberrypi/bookworm-feedback/issues/24

aallan, 2023-09-04:
Now a raspberrypi/bookworm-feedback issue.

closed 2023-09-04 by aallan

Vaxter, 2024-01-13:
> Reopening. Just tested the [Codec Zero](https://www.raspberrypi.com/documentation/accessories/audio.html#extra-codec-zero-configuration) part of the instructions, and the `sudo alsactl restore -f [...]

I have exactly the same issue and I have followed the documentation word for word.
When I run it in Desktop OS I get the error but the hat works, but if I take any headless version it's not working and recording produces a single crackling sound and shows overrun error with arecord.