raspberrypi/documentation #2939: Tutorial: Block ads on every device in your house with a Raspberry Pi and Pi-hole

tobybroberts, 2023-05-16:
Pi-hole using Raspberry Pi Bookworm OS fails to install. Currently Pi-Hole only support  Buster and Bullseye

aallan, 2023-07-30:
Still a thing.

aallan, 2023-09-04:
@nathan-contino Another Wordpress one for you!

aallan, 2023-09-20:
So Pi Hole now supports Debian 12 (aka Bookworm) according to https://docs.pi-hole.net/main/prerequisites/#supported-operating-systems. While it doesn't officially support our downstream Bookworm on Raspberry Pi, there is presumably no show-stoppers here. Upstream is *probably now ready* for us to proceed on this one.

If you do run into problems with installation @nathan-contino, can you please raise a compatibility issue on the project's Github repo, https://github.com/pi-hole/pi-hole. If they're not preparing for Bookworm on Pi, they need to be.

nathan-contino, 2023-10-11:
Updated tutorial is live now!

closed 2023-10-11 by nathan-contino