raspberrypi/documentation #2943: Remove or heavily edit the Python documentation

lurch, 2023-05-16:
The Raspberry Pi documentation doesn't feel like the right place to be teaching people about variables and lists in Python. https://www.raspberrypi.com/documentation/computers/os.html#python

So at the very _least_ we should get rid of the "Basic Python usage" section https://www.raspberrypi.com/documentation/computers/os.html#basic-python-usage

aallan, 2023-05-16:
I think just cut this entire section, https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/computers/os/using-python.adoc.

lurch, 2023-05-16:
The only reason I didn't _immediately_ say cut the entire section, was because it also mentions piwheels, which **is** Raspberry Pi-specific.  :man_shrugging:

PR #2945 (2023-05-17): Remove Python-usage instructions
This is covered elsewhere, and not specific to the Raspberry Pi.
Fixes #2943
commit 9f9edec2 (2023-05-17): Remove Python-usage instructions, as this is covered elsewhere, and not specific to the Raspberry Pi. Fixes #2943
closed 2023-05-17 by aallan