raspberrypi/documentation #2954: "Legacy OS" will be switching from Buster to Bullseye

lurch, 2023-05-17:
So any Legacy OS references will need to be checked and updated.

lurch, 2023-08-21:
@jehlers42 FYI these are some pages that likely need to be edited:
```console
$ git grep -iE "jessie|buster|bullseye|legacy os"
documentation/asciidoc/computers/camera/libcamera_apps_getting_help.adoc:* If you are using _Buster_ please upgrade to the latest OS, as `libcamera-apps` is no longer supported there.
documentation/asciidoc/computers/camera/libcamera_apps_getting_started.adoc:NOTE: On Raspberry Pi 3 and earlier devices running _Bullseye_ you need to re-enable _Glamor_ in order to make the X Windows hardware accelerated preview window work. To do this enter `sudo raspi-config` at a terminal window and then choose `Advanced Options`, `Glamor` and `Yes`. Finally quit `raspi-config` and let it reboot your Raspberry Pi.
documentation/asciidoc/computers/camera/libcamera_apps_getting_started.adoc:NOTE: Raspberry Pi 3 and older devices running _Bullseye_ may not by default be using the correct display driver. Refer to the `/boot/config.txt` file and ensure that either `dtoverlay=vc4-fkms-v3d` or `dtoverlay=vc4-kms-v3d` is currently active. Please reboot if you needed to change this.
documentation/asciidoc/computers/camera/libcamera_python.adoc:Picamera2 is only supported on Raspberry Pi OS Bullseye (or later) images, both 32- and 64-bit.
documentation/asciidoc/computers/camera/libcamera_python.adoc:. Images based on Buster or earlier releases.
documentation/asciidoc/computers/camera/libcamera_python.adoc:. Bullseye images where the legacy camera stack has been re-enabled.
documentation/asciidoc/computers/configuration/access-point-bridged.adoc:This documentation was tested on a Raspberry Pi 3B running a fresh installation of Raspberry Pi OS Buster.
documentation/asciidoc/computers/configuration/access-point-routed.adoc:This documentation was tested on a Raspberry Pi 3B running a fresh installation of Raspberry Pi OS Buster.
documentation/asciidoc/computers/configuration/configuring-networking.adoc:Note that with the latest Buster Raspberry Pi OS release, you must ensure that the `wpa_supplicant.conf` file contains the following information at the top:
documentation/asciidoc/computers/legacy_config_txt/memory.adoc:IMPORTANT: The default camera stack (libcamera) on Raspberry Pi OS - Bullseye uses Linux CMA memory to allocate buffers instead of GPU memory so there is no benefit in increasing the GPU memory size.
documentation/asciidoc/computers/os/updating.adoc:Generally speaking, doing this regularly will keep your installation up to date for the particular major Raspberry Pi OS release you are using (e.g. Buster). It will not update from one major release to another, for example, Stretch to Buster or Buster to Bullseye.
documentation/asciidoc/computers/os/updating.adoc:The latest version of Raspberry Pi OS is based on https://www.raspberrypi.com/news/raspberry-pi-os-debian-bullseye/[Debian Bullseye]. The previous version was based on https://www.raspberrypi.com/news/buster-the-new-version-of-raspbian/[Buster]. If you want to perform an in-place upgrade from Buster to Bullseye (and you're aware of the risks) see the https://forums.raspberrypi.com/viewtopic.php?t=323279[instructions in the forums].
documentation/asciidoc/computers/raspberry-pi/boot-nvme.adoc:You must have the latest versions of the VideoCore firmware and Raspberry Pi OS Linux kernel to boot directly from an NVMe SSD disk. The Raspberry Pi Bullseye and Buster Legacy releases have everything needed.
documentation/asciidoc/computers/raspberry-pi/display-parallel-interface.adoc:Display Parallel Interface (DPI) displays can be connected to Raspberry Pi devices via the 40-pin general-purpose input/output (GPIO) connector as an alternative to using the dedicated Display Serial Interface (DSI) or High-Definition Multimedia Interface (HDMI) ports. Many third-party DPI displays have been made available to take advantage of this. The Buster (and earlier) Raspberry Pi operating system (OS) and the legacy display stack used Raspberry Pi-specific parameters in config.txt to configure DPI displays. With the move to Bullseye and its use of the Kernel Mode Setting (KMS) graphics driver by default, these config.txt entries are no longer relevant as all control of the display pipeline has shifted to the Linux kernel.
documentation/asciidoc/computers/remote-access/network-boot-ipv6.adoc:To boot via IPv6 you need an updated version of the firmware (e.g. `start4.elf`) and the bootloader. Using the Bullseye release of Raspberry Pi OS and the latest stable bootloader should be sufficient.
```

The current situation is that "Buster" is the Legacy OS and "Bullseye" is the current OS https://www.raspberrypi.com/software/operating-systems/ but after we release Raspberry Pi OS Bookworm, "Bullseye" will then be the Legacy OS, and "Bookworm" will be the current OS. (And Buster will be unsupported, so we probably shouldn't mention it in the docs?)

aallan, 2023-08-24:
We have merged the thing, so yes!