raspberrypi/documentation #2955: raspi-config options have changed

lurch, 2023-05-18:
(and may still be changing)
https://www.raspberrypi.com/documentation/computers/configuration.html#list-of-options

lurch, 2023-05-24:
@spl237 tells me that the contents of the "Advanced" menu will vary depending on which model of Pi is being used, and also on which graphics stack is being used. He says that "most users" shouldn't be changing these settings, and therefore we shouldn't mention them in our documentation. The only options of raspi-config that we ought to be documenting are those which are also displayed in the "GUI version" i.e. the "Raspberry Pi Configuration" program.

aallan, 2023-05-24:
I use the Advanced Menu pretty much every time I set up a Pi. It will be documented.

spl237, 2023-05-24:
You shouldn't be using it. What are you doing in the Advanced Menu that needs doing every time you set up a Pi?

aallan, 2023-07-30:
Still a thing.

jehlers42, 2023-08-08:
@spl237 My PR'd changes to the getting started guide has a quick mention of the menu, basically calling it out before the user would hit "Write": "Optionally, you can click the cog wheel icon to open the <<Advanced options>> menu."

We also call out a few times about the need to use the advanced menu to create users for headless setups. Granted I'm still new to this whole Pi thing so I've mostly been going with what already existed in the guide so I'm not sure how needed that info truly is.

I do tend to agree that "service only" type situations shouldn't be fully documented, but this one seems easily-exposed to our users since it's a simple cogwheel icon on the installation GUI. Curious users are likely to click it and potentially get themselves in trouble. Ideally the UX would be the first line of defence here, with a "are you sure you want to do this?" confirmation dialog or tooltip.

Otherwise my suggestion would be a warning box like you recommend with a link to a doc that does explain the menu, but also makes it very clear the risks associated with changing those settings.

aallan, 2023-08-08:
@jehlers42 I think you're talking about the Imager, we were talking about the raspi-config command line tool which is used post-installation.

spl237, 2023-08-25:
Yes, apologies for that - we hopefully won't be making many more changes, but the image is still displaying more churn than I would like...

aallan, 2023-09-04:
Okay. So the decision is to document `raspi-config` as it works on Pi 4, with a `NOTE` block to say that options in the Advanced menu will depend on the underlying hardware you're running things on and will change between boards.

@lurch assigning to you.

lurch, 2023-09-06:
Fixed on `bookworm` branch.

commit df443eb8 (2023-09-06): Update raspi-config documentation
Fixes #2955
closed 2023-09-06 by lurch