raspberrypi/documentation #2960: Various parts of camera-software page need to be updated

lurch, 2023-05-18:
see e.g.
https://www.raspberrypi.com/documentation/computers/camera_software.html#driver-differences-when-using-libcamera-or-the-legacy-stack
https://www.raspberrypi.com/documentation/computers/camera_software.html#software-interfaces
https://www.raspberrypi.com/documentation/computers/camera_software.html#raspicam-applications

aallan, 2023-05-18:
> https://www.raspberrypi.com/documentation/computers/camera_software.html#raspicam-applications

I think it's likely we'll cut the chord and remove documentation around the legacy raspicam apps.

lurch, 2023-05-18:
> I think it's likely we'll cut the chord and remove documentation around the legacy raspicam apps.

Yes, there's already #2938 for that; this issue is more about updating any "other wording" that refers to the legacy raspicam apps.

closed 2023-06-27 by lurch