raspberrypi/documentation #2961: Use `alsactl store` instead of `restore` the first time around

cmangla, 2023-05-18:
https://github.com/raspberrypi/documentation/blob/04d467db848430e4ee6496fd95484e58bf98dc26/documentation/asciidoc/accessories/audio/configuration.adoc?plain=1#L90

It seems the very first time around, one needs to run `alsactl store` instead of `alsactl restore`. If I run `restore` first, it just errors and doesn't seem to do anything. After using `store` initially, subsequent `restore` commands work.

Furthermore, I note that the `sudo` in the next instructions to setup `/etc/rc.local` is redundant. But, that command does work fine as long as initially you've used `alsactl store`

cmangla, 2023-05-18:
See https://forums.raspberrypi.com/viewtopic.php?p=2107971#p2107971 for details

cmangla, 2023-05-18:
It is also probably safer to add the `-D Zero` argument to all the `alsactl` commands.

aallan, 2023-05-19:
@tobybroberts Could you take a look?

cmangla, 2023-05-19:
Actually it seems the boot command also needs to be `alsactl store`

lurch, 2023-05-22:
> `sudo alsactl restore -f /home/pi/Pi-Codec/IQaudIO_Codec_OnboardMIC_record_and_SPK_playback.state`

Did you remember to clone the https://github.com/iqaudio/Pi-Codec.git repo? Obviously if you're using a username other than `pi`, you'll need to change the `/home/pi/` part of that path.

lurch, 2023-05-22:
Ah, sorry, I've just read your forum thread and I see you _have_ already done that! :laughing: (I should have done that before I started typing :facepalm: )

I **suspect** that when you change `restore` to `store` it's simply overwriting the `/home/pi/Pi-Codec/IQaudIO_Codec_OnboardMIC_record_and_SPK_playback.state` with your card's _current_ configuration, which is why you're not getting any error. But the card's "default configuration" doesn't produce any audio output, which is why the `restore state` command was needed in the first place.
Unfortunately those instructions were written by @tobybroberts using the "old" black audio cards, so there must be some change in the green post-rebranding audio cards which means that those original `.state` files are no longer working correctly.

cmangla, 2023-05-23:
> I **suspect** that when you change `restore` to `store` it's simply overwriting the `/home/pi/Pi-Codec/IQaudIO_Codec_OnboardMIC_record_and_SPK_playback.state` with your card's _current_ configuration

I just checked, and you are right. `alsactl store -f prev.state` just stores something in `prev.state`, which I assume is the pre-existing config. So then restoring it works, but is rather pointless.

I downloaded a fresh copy of the state files from the git repo and I still get an error on trying `alsactl restore`:
```
# alsactl restore -f CZ/IQaudIO_Codec_OnboardMIC_record_and_SPK_playback.state 
No state is present for card Zero
alsa-lib main.c:1014:(snd_use_case_mgr_open) error: failed to import hw:0 use case configuration -2
Found hardware: "RPi_Codec_Zero" "" "" "" ""
Hardware is initialized using a generic method
No state is present for card Zero
```

cmangla, 2023-05-25:
After days of searching through forums and banging my head on it, I managed to get sound out of the codec-zero. I needed to change `IQaudIOCODEC` to `Zero` in the `.state` files. This was an arbitrary guess, and something I've seen in any of the forum posts I could find on the web.

cmangla, 2023-05-26:
I sent the above pull request for the modified `.state` files. The instructions in the documentation here should be modified if that pull request is merged.

cmangla, 2023-05-26:
It would be useful to add the following command to docs for headless setup, for testing it:
```
aplay -vvv -D sysdefault  /usr/share/sounds/alsa/Noise.wav
```

petercwallis, 2023-07-03:
Yes, someone who doesn't know anything needs to go through the install process and rewrite the instructions with someone who knows all about  alsa on piOS sitting behind them.  I got it working on two different installs but seem to have done it in two different ways. It is certainly not plug_and_play as the documentation suggests.  Controlling the volume from the sound icon in the task bar does not work, and I still can't identify the name of the mixer in snd_mixer_selem_id_set_name(sid, mixer_name) in the [rot.c ](https://github.com/iqaudio/tools/blob/master/IQ_rot.c) example.

aallan, 2023-07-17:
Fixed by #3014?

lurch, 2023-07-17:
Still pending on an additional tweak to https://github.com/raspberrypi/Pi-Codec - just waiting for @tobybroberts to be given write-access to that repo.

aallan, 2023-09-17:
Did this ever get resolved? @tobybroberts?

tobybroberts, 2023-09-17:
@aallan - Yes. A number of issues on this one. Overlooked this one. All sorted.

closed 2023-09-17 by aallan