raspberrypi/documentation #2966: libcamera_hello tutorial required

petercwallis, 2023-05-24:
The code for the libcamera apps is great and perfectly understandable, and I can modify it in situ, but cpp error messages!  I am having trouble writing a stand-alone c program that uses installed headers and libs. The use-case is, I want to be able to put the code in an arbitrary directory.  I have just used git, ninja, cmake, pip3, meson, and of course make to build the thing, and it seems I have to understand each(?) of those in order to create and run my own app.  It is also not clear how to add a post-processor - I can successfully modify the negative example, but cannot enable a differently named post-processor based on a copy of that code.

Could the tutorial provide a stand alone libcamera_hello.cpp and (hopefully at most) a Makefile to compile it?

lurch, 2023-05-24:
ping @davidplowman

petercwallis, 2023-07-03:
Here is what I was after:
https://github.com/petercwallis/libcamera

closed 2023-07-03 by petercwallis