raspberrypi/documentation #2973: Integration of White Papers into mainline documentation

aallan, 2023-06-02:
We need to properly integrate the White Papers into the mainline documentation.

The first thing here is to do another pass through the existing white papers to see if there is anything that should be surfaced more generally, and therefore needs to get poured into the documentation rather than exist as a separate document.

Next however is adding an info block to the online documentation to point users at white papers that extend the documentation. All the white papers should (in theory) be extensions — deep dives into certain topics — extending the existing documentation. We should therefore insert info blocks at the correct places in the documentation pointing users to that extended docuemntation.

I'm sort of envisioning a new type of [admonitions](https://docs.asciidoctor.org/asciidoc/latest/syntax-quick-reference/#admonitions)-style block for emphasis called `[WHITEPAPER]` which might look something like this,

[image]

although if we want to do an initial fast pass where we can start by using a standard `[INFO]` block to get them all integrated more quickly.

My understanding is that we can't add a new admonition-style block, so we'd probably have to add a macro and handle styling in pre-processing. I'd suggest it would look like this,

````
[whitepaper, title="White Paper Title", subtitle="A subtitle for the white paper", link=https://url.here]
----
Lorem ipsum dolor sit amet, consectetur adipiscing elit. Phasellus elementum purus a eros laoreet, a vestibulum risus interdum. Suspendisse bibendum sagittis felis et pulvinar. Fusce purus ante, vestibulum in lacinia a, placerat quis libero. Nunc nibh mauris, finibus vel fermentum vulputate, convallis id neque. In in rhoncus urna. Aliquam sed sapien porta, venenatis massa vitae, aliquam mauris. Nunc ut ante pharetra, faucibus ante convallis, venenatis ligula. 

Nullam volutpat, sem quis commodo condimentum, libero ex aliquam urna, in aliquet justo felis mattis libero. Pellentesque vestibulum enim vel augue blandit tempus. Ut a elit non turpis viverra aliquam sed a felis. Etiam ut porta mi. Sed feugiat arcu leo, vulputate placerat ante ullamcorper at. In id metus sit amet massa molestie efficitur. Morbi rutrum scelerisque metus id dignissim.
----

where the title, subtitle and link are automagically inserted into the text in the block.

aallan, 2023-06-02:
Ping @nelliemckesson and @lurch and @JamesH65.

First thing to do would be to get a list of all published White Papers in PIP and see where they extend the existing online documentation (@JamesH65) so we can insert an appropriate `[INFO]` block directing users to the extended docs.

Next thing to do would be @nelliemckesson and @lurch taking a look at how to implement the styling of a new block. Sounds like a @nelliemckesson thing in the first instance there.

aallan, 2023-06-02:
Maybe something like this would be cleaner, with the link being the title?

[image]

JamesH65, 2023-06-02:
The current list of WPs is here, https://pip.raspberrypi.com/categories/685-whitepapers-app-notes

JamesH65, 2023-06-02:
> Maybe something like this would be cleaner, with the link being the title?
> 
> [image]

It would be cool if clicking on the image of the front page also follows the link.

JamesH65, 2023-06-02:
It may or may not be useful but all WPs have metadata in them with title and subtitle, so they could be extracted from the PDF rather than being retyped in the docs. 
[image]

nelliemckesson, 2023-06-06:
@aallan In the interest of good HTML semantics, I'd propose we use a sidebar with a custom role for these, like this:

```
[.whitepaper, title="White Paper Title", subtitle="A subtitle for the white paper", link=https://url.here]
****
Lorem ipsum dolor sit amet, consectetur adipiscing elit. Phasellus elementum purus a eros laoreet, a vestibulum risus interdum. Suspendisse bibendum sagittis felis et pulvinar. Fusce purus ante, vestibulum in lacinia a, placerat quis libero. Nunc nibh mauris, finibus vel fermentum vulputate, convallis id neque. In in rhoncus urna. Aliquam sed sapien porta, venenatis massa vitae, aliquam mauris. Nunc ut ante pharetra, faucibus ante convallis, venenatis ligula. 

Nullam volutpat, sem quis commodo condimentum, libero ex aliquam urna, in aliquet justo felis mattis libero. Pellentesque vestibulum enim vel augue blandit tempus. Ut a elit non turpis viverra aliquam sed a felis. Etiam ut porta mi. Sed feugiat arcu leo, vulputate placerat ante ullamcorper at. In id metus sit amet massa molestie efficitur. Morbi rutrum scelerisque metus id dignissim.
****
```

But whichever block type we choose, adding a custom role would be really helpful.

Jekyll/asciidoctor have built-in handling for custom templates for specific elements or blocks. I threw together [an initial draft here](https://github.com/raspberrypi/documentation/pull/2975).

The sidebar markup above plus the changes in the PR linked above generate output like this (I'm auto-generating the cover image here -- all the CSS could use a little fine-tuning but I wanted to show you a rough idea of what we could do):
 
[image]

aallan, 2023-06-11:
> I'd propose we use a sidebar with a custom role for these, like this…

This works for me. If I merge the draft PR this is what we get? Although I have a feeling the CSS will inevitably need to be tweaked, I'm pretty happy with the above. Tempted to merge and get the published White Papers properly integrated into the docs this week. Are you happy for me to go ahead and merge @nelliemckesson ?

aallan, 2023-06-11:
> It may or may not be useful but all WPs have metadata in them with title and subtitle, so they could be extracted from the PDF rather than being retyped in the docs.

Huh. Nice idea, but a bit worried by adding too much magic in at this stage.

aallan, 2023-06-11:
> It would be cool if clicking on the image of the front page also follows the link.

Is this doable @nelliemckesson ?

nelliemckesson, 2023-06-12:
@aallan yeah should be doable. I’ll add that today and then we can merge!

aallan, 2023-06-12:
…and merged #2975

aallan, 2023-06-12:
I'll have a poke around and get a test case WP inline in the documentation tomorrow. Thanks @nelliemckesson.

aallan, 2023-06-13:
Hey @nelliemckesson. Can you add a bit more margin below white paper box, the following text seems to be a bit crushed up to it in my initial experimentations with things?

[image]

aallan, 2023-06-13:
Maybe just throw that change directly onto develop?

aallan, 2023-06-13:
…and fixed in #2981

aallan, 2023-06-14:
Looks like there is a problem with styling on production.

The white paper info boxes work in the local browser version, but don’t render correctly on the production server.

e.g. see https://raw.githubusercontent.com/raspberrypi/documentation/develop/documentation/asciidoc/computers/raspberry-pi/raspberry-pi-industrial.adoc which has a white paper block at the top.

This renders like this locally,

[image]

But like this on production,

[image]

aallan, 2023-06-14:
Ping @nelliemckesson 👆

aallan, 2023-06-14:
Ping also @mudge who might know why it builds locally but not on production.

mudge, 2023-06-14:
@aallan FWIW it currently looks like this for me on production:

[image]

aallan, 2023-06-14:
Caching?

aallan, 2023-06-14:
Thanks @mudge, that makes sense!

lurch, 2023-06-15:
>we didn't forcibly bust any browser caches by incrementing the ver number. In future, if we modify that URL, it should force everyone's browsers to fetch the latest styles.

@mudge Given the pre-processing that we do when building the production site, would it _perhaps_ make sense to do something like:
```
<link rel="stylesheet" href="{{ site.baseurl }}/css/style.css?ver={{ md5sum.style.css }}">
```

(there's no plumbing in place for this yet, it's just a suggestion)

mudge, 2023-06-15:
@lurch that'd be ideal (we typically "fingerprint" assets like this across raspberrypi.com).

lurch, 2023-06-16:
> @lurch that'd be ideal

I've created #2991