raspberrypi/documentation #2974: OpenSourcing manuals to fix issues

FelixSelter, 2023-06-06:
Hi,

I thought I would ask this question here as there seems to be no place it fits.  
The current manuals bcm 2711 peripherals and most likely others contain some issues.  
If they would be open sourced we could help fixing them.

aallan, 2023-06-06:
Our [online documentation](https://www.raspberrypi.com/documentation/) is published under a Creative Commons Attribution-ShareAlike 4.0 International (CC BY-SA) licence, and the source along with the toolchain to build the documentation site is available [on Github](https://github.com/raspberrypi/documentation/). We welcome issues and PRs.

For the most part our PDF-based documentation is released under a Creative Commons Attribution-NoDerivatives 4.0 International (CC BY-ND) license, although the BCM2711 peripherals document that you mentioned is not, as it is based on previous documentation published by Broadcom, and all rights are reserved to the respective copyright holders.

We are not anticipating changing the current CC BY-ND license, or releasing the source or toolchain to build the PDF documentation at any point in the future, although we're happy to hear feedback and issues either here or for the RP2040 documentation over on the [pico-feedback](https://github.com/raspberrypi/pico-feedback/issues) Github repo.

closed 2023-06-06 by aallan

aallan, 2023-06-07:
We have made public everything we can based on our agreements with Broadcom. 

There are [a couple of issues](https://github.com/raspberrypi/documentation/issues?q=is%3Aissue+is%3Aopen+bcm2711) in backlog with respect to the BCM2711 peripherals document, they'll get updated when there is a gap in the schedule for someone to resolve them.