raspberrypi/documentation #2977: Ambiguity about powering the Touch Display

lurch, 2023-06-07:
@jackbenwillis noticed that when connecting the Touch Display to a Raspberry Pi, you then have two micro-USB power-input sockets (one on the Pi, one on the Display) and https://www.raspberrypi.com/documentation/accessories/display.html doesn't explain which one you should use. He asked some engineers, and got told "you can use either, but because Pi 4 requires more power the official line should be to power the Pi not the touch display", so we should reflect that in the documentation.

aallan, 2023-06-07:
I just looked at the [the documentation](https://www.raspberrypi.com/documentation/accessories/display.html) on the Touch Display and it needs totally overhauling. The page is a bit of a relic.

aallan, 2023-08-15:
Still needs doing.

closed 2023-09-25 by aallan