raspberrypi/documentation #2987: Typo in DPI documentation (horizontal vs vertical)

toptensoftware, 2023-06-15:
Shouldn't this say "horizontal" back padding (not "vertical").

https://github.com/raspberrypi/documentation/blob/dcbaf3813d7ff04e2bd2d4192a615597633070dc/documentation/asciidoc/computers/raspberry-pi/display-parallel-interface.adoc?plain=1#L426

lurch, 2023-06-30:
ping @popcornmix and @6by9

popcornmix, 2023-06-30:
The fact it says:
```
<h_back_porch>    = vertical back padding from DE active edge
...
<v_back_porch>    = vertical back padding from DE active edge
```
Does suggest the first is a typo.

6by9, 2023-06-30:
I'd be surprised if vertical syncs are with regard the DE active edge, but I can't find the relevant MIPI spec quickly to confirm.

lurch, 2023-06-30:
#2989 says "Use of dpi_timings= is only applicable to the legacy display stack" so closing this issue as invalid too.

closed 2023-06-30 by lurch