raspberrypi/documentation #2989: Clarification of pixel_freq in DPI docs

toptensoftware, 2023-06-15:
Does "width*height" mean active width/height or width/height including front/back porch and sync pulses?  Normally it would include these extra values.  Either way, it should be clearly stated.

https://github.com/raspberrypi/documentation/blob/dcbaf3813d7ff04e2bd2d4192a615597633070dc/documentation/asciidoc/computers/raspberry-pi/display-parallel-interface.adoc?plain=1#L437

6by9, 2023-06-30:
pixel_freq is 
```
(h_active_pixels + h_front_porch + h_sync_pulse + h_back_porch) * (v_active_lines + v_front_porch + v_sync_pulse + v_back_porch) * framerate
```

Use of ```dpi_timings=``` is only applicable to the legacy display stack, therefore there won't be further work on it.

6by9, 2023-06-30:
FWIW, The equivalent to dpi_timings is now the vc4-kms-dpi-generic overlay.
See https://github.com/raspberrypi/linux/blob/rpi-6.1.y/arch/arm/boot/dts/overlays/README#L4380-L4432
Clock configuration is all through the standard Linux clock frameworks.

lurch, 2023-06-30:
> FWIW, The equivalent to dpi_timings is now the vc4-kms-dpi-generic overlay.

Reopening this and adding it to the Bookworm milestone.

JamesH65, 2023-06-30:
There are KMS DPI instructions here 

https://pip.raspberrypi.com/categories/685-whitepapers-app-notes/documents/RP-003467-WP/Transitioning-from-Buster-to-Bullseye.pdf

 and also some useful data here

https://pip.raspberrypi.com/categories/685-whitepapers-app-notes/documents/RP-003471-WP/Using-a-DPI-display.pdf

toptensoftware, 2023-06-30:
Two questions:

1. I'm still interested in some documentation on what pixel clock frequencies are supported (see original post on this issue)
2. Are the `dpi_timings` settings being superceded, removed and replaced by \vc4-kms-dpi-generic? Or will those setting be available going forward for non-linux environments (eg: bare metal)?

aallan, 2023-09-04:
Still needs doing.

aallan, 2023-09-06:
Last issue left that will let us close one of the `bookworm` mother issues. @nathan-contino can you look at this next? Thanks!

aallan, 2023-09-12:
@nathan-contino Can you knock this one off next if possible?

closed 2023-09-14 by aallan