raspberrypi/documentation #2991: Add cache-busting fingerprints

lurch, 2023-06-16:
As discussed in #2973 , the caching used for CSS files, JS files etc. means that whenever we change one of these file-types and then "push to production", it can be up to 20 minutes or so before things "sync up again", and during this time people might be seeing "incorrect" content.
This can be fixed by adding a `?ver=1646652613` to the end of the file's URL (with a unique number each time), but as this is a manual process we'll probably forget to do this most of the time :wink:  However we could instead automate this by making the `ver` value e.g. the md5sum of the file's content.

@mudge If/when we do this, what file-types would we need to do this for? Is MD5 a suitable algorithm for this, or do you have another recommendation? (obviously this is only a content-has-changed checksum, not a cryptographically-secure verification! :joy: )

mudge, 2023-06-26:
@lurch I'd start with CSS and JS files (typically, asset pipelines will do this for image and font filenames too but that would require rewriting any references in CSS files so I don't think this is worth it for now).

Given the use case, MD5 should be fine though another option might be SHA1 (as used by [Rails' Propshaft](https://github.com/rails/propshaft/blob/26082d0b8a5106a79c7e0bbb65ef780cc47bb837/lib/propshaft/asset.rb#L29)).

lurch, 2023-08-27:
This could/should still be done at some point, but it's currently _very_ low down on the priority-list!

aallan, 2023-09-04:
This sounds like a combination of @nelliemckesson and @mudge maybe?

closed 2023-09-05 by aallan