raspberrypi/documentation #2993: Purge mentions of "X Windows" etc.

lurch, 2023-06-26:
Some of the documentation refers to "X Windows" or "X11" as a way to differentiate the "Desktop" version of Raspberry Pi OS from the "Lite" (headless) version of Raspberry Pi OS. However with the introduction of Wayland in Bookworm, it doesn't make sense to refer specifically to X, so we should just use generic phrasing like "graphical environment".

aallan, 2023-06-26:
> However with the introduction of Wayland in Bookworm, it doesn't make sense to refer specifically to X, so we should just use generic phrasing like "graphical environment".

Use "desktop" or "graphical environment" I would think.

JamesH65, 2023-06-26:
The download page and Imager use "with desktop" or "with Raspberry Pi Desktop" respectively. And then Imager uses "desktop environment" on the next page.

aallan, 2023-06-26:
That's probably the answer then.

lurch, 2023-06-26:
Okay, so are we all happy with "desktop environment" then?

jackbenwillis, 2023-06-26:
We ought to ask Brian O about this, but I think "Raspberry Pi Desktop" refers to this:
https://www.raspberrypi.com/software/raspberry-pi-desktop/

So probably best to avoid calling the Raspberry Pi OS desktop environment "Raspberry Pi Desktop"

brian-rpi, 2023-06-27:
Yup, Raspberry Pi Desktop is its own thing (not 100% what the status of it is). "Desktop environment" is probably the safest way to go across everything, though if we wanted to hammer home in Imager that we're not using a standard Debian environment we could specify "a Raspberry Pi-designed desktop environment" but I'm pretty agnostic

jehlers42, 2023-08-08:
Some of the references to X Windows also mention Qt. For example, "The only difference is that the latter also builds the `qcam` test application, which has dependencies on Qt and X Windows"

Assuming the Qt dependency still exists, does it make sense in these cases to just remove the X reference instead of replacing it with "desktop environment"?

lurch, 2023-08-09:
@jehlers42 Yeah, that's [libcamera-apps](https://github.com/raspberrypi/libcamera-apps) stuff, which I'll be discussing with @davidplowman and @naushir when I get to updating that part of the Bookworm-docs. We'll also need instructions from them for how to [build things](https://www.raspberrypi.com/documentation/computers/camera_software.html#building-libcamera-and-libcamera-apps) without using `pip`.

jehlers42, 2023-08-14:
PR for most of the easy changes: https://github.com/raspberrypi/documentation/pull/3049

@lurch, any references outside of this PR likely need your knowledge to modify.

lurch, 2023-08-30:
This has now been complicated slightly by #3057

lurch, 2023-09-12:
Finished checking the docs, and I don't think there's any more "purging" needed, due to #3057

closed 2023-09-12 by lurch