raspberrypi/documentation #2996: /opt/vc/ directory no longer exists

lurch, 2023-06-28:
...so any mentions of it should be deleted, or edited as appropriate.

jehlers42, 2023-08-08:
Is there a replacement or alternate directory the docs should reference?

Looks like most of the references are related to the command line. For example:
* omxplayer --display n /opt/vc/src/hello_pi/hello_video/test.h264
* cc myfile.c -I/opt/vc/include -L/opt/vc/lib -lbcm_host -o myfile
* Enter the following command to give a list of DMT-supported modes: `/opt/vc/bin/tvservice -m DMT`

lurch, 2023-08-08:
I've already done parts of this (on the [bookworm](https://github.com/raspberrypi/documentation/tree/bookworm) branch), and other parts will get done in due course, so I'm going to assign this back to me :slightly_smiling_face:

lurch, 2023-08-30:
On the bookworm branch, `/opt/vc/` now only appears in `documentation/asciidoc/computers/legacy_config_txt/video.adoc` so I think this can be closed now.

closed 2023-08-30 by lurch