raspberrypi/documentation #2999: Updated screenshots

lurch, 2023-07-04:
Some of the screenshots on different pages will need updating for Bookworm - especially any screenshot that shows the old "orange" default Bullseye desktop-background instead of the new "blue" Bookworm desktop-background.
This is a meta-issue for listing all the screenshots that need updating, and will be added to over time.

lurch, 2023-07-12:
Images used on https://www.raspberrypi.com/documentation/accessories/build-hat.html#setting-up-your-raspberry-pi (the majority of the images on this page are still okay)
* https://www.raspberrypi.com/documentation/accessories/images/setting-up.png
* https://www.raspberrypi.com/documentation/accessories/images/raspi-config-2.png

aallan, 2023-08-21:
@jehlers42 Can you walk through the entire documentation and find out how many screenshots we actually have (that are used?)

jehlers42, 2023-08-22:
Finished preliminary audit and attached the results. Organised by section, sub-section, and sub-heading.

[imageAudit.md](https://github.com/raspberrypi/documentation/files/12408451/imageAudit.md)

lurch, 2023-08-22:
Thanks @jehlers42 

145 images in total
```console
$ grep images/ ~/Downloads/imageAudit.md | wc -l
145
```

57 in the "Computers" section
```console
$ grep computers/images/ ~/Downloads/imageAudit.md | wc -l
57
```

aallan, 2023-08-22:
Just need the screenshots, only ones that need updated for Bookworm!

jehlers42, 2023-08-22:
File reuploaded with screenshots only.

[imageAudit.md](https://github.com/raspberrypi/documentation/files/12408601/imageAudit.md)

aallan, 2023-08-22:
Okay, 33 screenshots need updating for `bookworm`

aallan, 2023-08-22:
Okay. A lot of these don't need updating after all. Despite being screenshots. Only ones that need updating are:

```
            https://www.raspberrypi.com/documentation/computers/images/raspi-config.png

            https://www.raspberrypi.com/documentation/computers/images/wifi2.png
            https://www.raspberrypi.com/documentation/computers/images/key.png

            https://www.raspberrypi.com/documentation/accessories/images/setting-up.png
            https://www.raspberrypi.com/documentation/accessories/images/raspi-config-1.png
            https://www.raspberrypi.com/documentation/accessories/images/raspi-config-2.png
            https://www.raspberrypi.com/documentation/accessories/images/raspi-config-3.png
            https://www.raspberrypi.com/documentation/accessories/images/raspi-config-4.png
            https://www.raspberrypi.com/documentation/accessories/images/raspi-config-5.png

            https://www.raspberrypi.com/documentation/accessories/images/gui.png

            https://www.raspberrypi.com/documentation/accessories/images/firmware-update/warning.png
            https://www.raspberrypi.com/documentation/accessories/images/firmware-update/select.png
            https://www.raspberrypi.com/documentation/accessories/images/firmware-update/flashing.png
            https://www.raspberrypi.com/documentation/accessories/images/firmware-update/flashed.png
```

Even then I think some of them only needed checked to make sure they're the same, or haven't changed. The only *definitely need to be updated* ones are:

```
            https://www.raspberrypi.com/documentation/computers/images/wifi2.png
            https://www.raspberrypi.com/documentation/computers/images/key.png
            https://www.raspberrypi.com/documentation/accessories/images/gui.png
```

So just 3 images!

lurch, 2023-08-22:
Although _some_ of those screenshots might not need re-taking if they already look the same as they do on Bookworm.

aallan, 2023-08-22:
Oh. Also these,

```
https://www.raspberrypi.com/documentation/accessories/images/setting-up.png
https://www.raspberrypi.com/documentation/accessories/images/raspi-config-2.png
```

so 5.

aallan, 2023-09-04:
Okay, @jehlers42 can you go through and take the appropriate screenshots for this one? Pi 4 running latest Bookworm.

lurch, 2023-09-06:
https://github.com/raspberrypi/documentation/blob/bookworm/documentation/asciidoc/computers/configuration/images/raspi-config.png needs updating too.

aallan, 2023-09-06:
> https://github.com/raspberrypi/documentation/blob/bookworm/documentation/asciidoc/computers/configuration/images/raspi-config.png needs updating too.

For clarity @jehlers42. New screenshots shouldn't look like that, can we have window/terminal screenshots that looks like this please,

[image]

Stylistically we're looking for something closer to https://www.raspberrypi.com/documentation/accessories/build-hat.html or https://www.raspberrypi.com/documentation/microcontrollers/debug-probe.html.

aallan, 2023-09-06:
While the screenshot at the top of https://www.raspberrypi.com/documentation/computers/configuration.html#the-raspi-config-tool is being updated, it'd be good to add screenshots showing each of the top level categories to break the (very long) list that follows it up a bit. List is pretty impenetrable right now.

jehlers42, 2023-09-11:
@aallan branch is up with one image so far just to check and see if this is what you were hoping for re: new raspi-config images

https://github.com/raspberrypi/documentation/compare/bookworm...iss-2999?expand=1

I've also got each of the sub-menus ready to add.

aallan, 2023-09-12:
Yup, that looks fine!

aallan, 2023-09-13:
So #3109 closed this one? Right?

lurch, 2023-09-13:
Not quite. @jehlers42 added some _new_ images in #3109 , but he didn't edit the asciidoc to link them in yet...

closed 2023-09-14 by aallan