raspberrypi/documentation #3000: Add cables section to Accessories tab

aallan, 2023-07-07:
We have some [new cables](https://thepihut.com/products/official-raspberry-pi-micro-hdmi-to-vga-cable?mc_cid=836758cb25&mc_eid=fbb9223795), we potentially need a cables and dongles section in the documentation?

aallan, 2023-08-24:
Guess this should get done around the same time as we do #3080 perhaps?

jehlers42, 2023-09-18:
@aallan I'm assuming the intent here is to link to as many of our official products as possible, yes? For items that are official (or recommended?) that aren't on our own branded storefront, is there a retailer I should default to?

lurch, 2023-09-19:
I may be wrong, but I believe the _intention_ is that each of our official products should (eventually) have its own page at https://www.raspberrypi.com/products/ (although some pages e.g. https://www.raspberrypi.com/products/raspberry-pi-pico/ cover multiple different products).

aallan, 2023-09-25:
@nathan-contino We should sit down with Mike in Commercial before this one gets done, something for October.