raspberrypi/documentation #3003: Check for changes around syslog?

aallan, 2023-07-10:
So it looks like `rsyslog` is no longer enabled by default in Debian Bookworm. This means no /var/log/{syslog,messages}, instead you have to `sudo journalctl`.

We might have decided to renable it, or we might have left it disabled. Need to check and make changes if so.

lurch, 2023-07-10:
I asked @XECDesign about this a while ago, and IIRC he said "we just do whatever upstream Debian does" ?

aallan, 2023-07-10:
Right, need to make sure this gets fixed up then if it comes up. But I can't believe that syslog hasn't been mentioned at all in the documentation at some point.

lurch, 2023-07-10:
`syslog` isn't mentioned, but `/var/log` is mentioned in:
computers/configuration/securing-the-raspberry-pi.adoc
computers/remote-access/network-boot-raspberry-pi.adoc
computers/remote-access/secure-shell-passwordless.adoc

aallan, 2023-07-10:
Right so all of those references need updated to use `sudo journalctl`

lurch, 2023-08-17:
Some info in https://forums.raspberrypi.com/viewtopic.php?t=355066

aallan, 2023-08-17:
> Some info in https://forums.raspberrypi.com/viewtopic.php?t=355066

Doesn't seem to have much beyond my initial post, although it does confirm that `journalctl` is the default now and we aren't switching back. So any reference to `/var/log` in the docs needs updating.

closed 2023-09-14 by aallan