raspberrypi/documentation #3007: Touch Display installation needs better instructions

lurch, 2023-07-11:
The photo at https://www.raspberrypi.com/documentation/accessories/display.html#physical-installation doesn't seem very clear as to exactly which pins are being connected to which. Especially given that the pin-labels on the Touch Display's PCB aren't readable once a Raspberry Pi has been screwed onto the standoffs!

We should also add some wording about "if the touchscreen printed circuit board is powered directly", similar to that shown in https://datasheets.raspberrypi.com/display/7-inch-display-product-brief.pdf

aallan, 2023-08-25:
See also #3007

aallan, 2023-09-25:
So #3121 and subsequent changes by me to fix the narrative have fixed some of the problems here, but it needs an actual installation walkthrough, with step-by-step pictures. 

Possibly we should use GIFs like the [Build HAT instructions](https://www.raspberrypi.com/documentation/accessories/build-hat.html#preparing-your-build-hat) for this. That'll need Brian or possibly Jack involved.

aallan, 2023-09-25:
One for you @nathan-contino, but networking things first!

nathan-contino, 2024-06-06:
The installation instructions have since been overhauled significantly. I think this has been addressed!

closed 2024-06-06 by nathan-contino