raspberrypi/documentation #3013: NetworkManager randomised MAC addresses

aallan, 2023-07-14:
Document NetworkManager randomising MAC addresses, see for instance https://forums.raspberrypi.com/viewtopic.php?t=353678.

lurch, 2023-07-14:
Latest comment in that forum thread suggests that MAC address randomisation might get disabled by default, in which case no change would be needed to the docs.

aallan, 2023-07-16:
Need to check with @ghollingworth if there's been a final decision there then.

ghollingworth, 2023-07-17:
Not sure this needs documenting?

It's normal behaviour for WiFi devices to randomise their MAC address while scanning for APs.

aallan, 2023-07-17:
> It's normal behaviour for WiFi devices to randomise their MAC address while scanning for APs.

My impression was that the randomization was in a phone-like way every time it connected to the network, rather than just during the scanning stage?

lurch, 2023-08-03:
According to that same forum thread, this has now been disabled. Closing this issue.

closed 2023-08-03 by lurch