raspberrypi/documentation #3017: autoboot.adoc does not sufficiently explain the auto/tryboot mechanism

beartronix, 2023-07-24:
Hello,
sorry if I'm making any formal mistakes. This is my first Issue report.

My problems with [autoboot.adoc](https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/computers/config_txt/autoboot.adoc) are:

1. TRYBOOT link is dead (raspberry-pi.adoc does not exist anymore)
2. Why is there no connection to cmdline.txt? after all, the root partition selection happens there! How can it not have an influence on tryboot?
3. How does the partition numbering in autoboot.txt work? does it start with 0 or 1?

In autoboot.adoc, it would be nice to see the partition layout of the "example" code (boot_partitions 2 and 3).

I assume I never got into tryboot mode (only trying secondary rootfs for now, with tryboot_a_b=1); the files in `/proc/device-tree/chosen/bootloader` are empty.

Is it possible to complement autoboot.adoc with information how it connects to other mechanisms in the boot process (and answers to my questions above)?

Thank you for reading and best regards,
Stefan

aallan, 2023-08-25:
@jehlers42 can you follow up on this one? Thanks!

stevedavies-rp, 2026-08-27:
Many thanks for your patience. Your questions and suggestions are now being addressed in a documentation edit.

closed 2026-08-27 by stevedavies-rp