raspberrypi/documentation #3019: Instructions to build a kernel fail unless KERNEL is exported

mtaunton, 2023-07-24:
I carefully followed https://www.raspberrypi.com/documentation/computers/linux_kernel.html#building-the-kernel-locally, to build a 32-bit kernel from scratch for use in developing a device driver for my RPi400. However, building the kernel  (make -j4 zImage modules dtbs) failed very quickly, complaining of an unknown target "zImage".

I eventually found the cause to be that while KERNEL was correctly defined as a shell variable, it was not being exported, so make did not see it in its environment.  The fix is simply to prefix the KERNEL= definition with "export ".

I have created a branch with the relevant lines corrected.

lurch, 2023-07-25:
Duplicate of #2765 ?

Note also that the Pi 400 (and Pi 4, CM4, etc.) now default to booting a 64-bit kernel, even if you're running a 32-bit version of Raspberry Pi OS. https://www.raspberrypi.com/documentation/computers/config_txt.html#arm_64bit

aallan, 2023-07-25:
From @pelwell's comments in #2765 it doesn't look like the `export` line is needed?

pelwell, 2023-07-25:
@mtaunton Please post the build commands you used from your `history` - I'm skeptical that you have followed the instructions to the letter.

mtaunton, 2023-07-26:
@pelwell  I don't have that history available now. So I'm currently trying to replicate the process from scratch to confirm (or disconfirm) that it's a real problem. Unfortunately I'm now hitting a different problem. Will report back later.

mtaunton, 2023-07-26:
@pelwell  I did follow the instructions to the letter, and I did experience the problem as reported. However my explanation of what was wrong and my proposed solution are not correct. 

I now believe the problem's root cause is related to what @lurch mentioned, and involves an unexpected (and to my mind, undesirable) dependency of the local kernel build process on the version of the kernel that the build is run on. Here's a summary of the history:

I was using a Pi 400 running a freshly downloaded 32b Raspberry Pi OS. None of the instructions that I had followed mentioned that the default kernel running in this case is actually 64b, despite the system being "32b", so I was unaware of that. 

I wanted to build a 32b kernel to run on the same machine and also a 2GB Pi 4B (the target for my current development), neither of which need a 64b kernel. So I followed the instructions which say to set KERNEL=kernel7l before doing "make bcm2711_defconfig". I then customised my build version in CONFIG_LOCALVERSION - though I was puzzled why it had "v8" in the default value - by suffixing it with "-MT". When I then I typed "make -j4 zImage", the output was (just as I reported above):

```
  SYNC    include/config/auto.conf.cmd
make: *** No rule to make target 'zImage'.  Stop.
```

Subsequently, I found out about the 32b vs 64b kernel version detail that @lurch mentioned. So I switched to working under a 32b kernel (arm_64bit=0 in /boot/config.txt).  But being still puzzled about the "-v8" version I had seen for CONFIG_LOCALVERSION on the first attempt, I prefixed "KERNEL=kernel7l" with "export" on the assumption that that might fix the problem.  After running make bcm2711_defconfig and opening .config to customise it, the existing value was now "-v7l", so I supposed that the "export" had fixed the problem. That the kernel build also then succeeded further strengthened this conclusion, and hence I created a patch and submitted this issue / pull request.

However, after reading your comment above, further digging into the matter has revealed that, despite all appearances from the instructions, the setting of KERNEL has no effect at all on what kernel version is actually built! Rather, it seems the version is determined purely from the version of the kernel that the build is run **on**. 

Ideally (as I see it), the makery should be improved to allow building any target kernel version, independent of the kernel version the build is run under. But if not, then the instructions need an update to make the matter clear (you can only do a local build of an 32-bit kernel under a 32-bit kernel, and the same for 64-bit), or to make explicit that you must do such a build as a cross-compilation, not a "local" one, despite the host and target machines being the same. 

I hope that all makes sense?

pelwell, 2023-07-27:
A 64-bit kernel build on a 32-bit OS is effectively a cross-compile, so a note saying to jump to "Cross-Compiling the Kernel" in that situation should be sufficient.

mtaunton, 2023-07-27:
My problem was not building a 64-bit kernel on a 32-bit OS, but building a 32-bit kernel on a 32-bit OS, which does not look like a cross-compile. The trouble happened because despite the OS being 32-bit, the running kernel was actually 64-bit. Although it is now apparently the default, the instructions make no mention of this "mixed" case, hence the confusion.

pelwell, 2023-07-27:
In the absence of an explicit "ARCH=arm" or "ARCH=arm64", the kernel has to guess. The guesswork happens at the "make ... *_defconfig" stage, and it appears to follow the kernel version rather than the OS, i.e. with a 64-bit kernel on a 32-bit OS it will build a 64-bit kernel.

If you switched to explicitly running a 32-bit kernel _without_ rerunning the configuration step then it will still end up 
Following the cross-compilation instructions for the 64-bit kernel on removes the need for guesswork.

Whether you realised it or not, you _were_ attempting to build a 64-bit kernel on a 32-bit OS. My suggestion above remains unaltered.

XECDesign, 2023-07-27:
> ping @XECDesign I think you mentioned something about this to me a while ago?

Nope, nothing to do with me. I'd only add that a lot of software uses the running kernel architecture to determine the target architecture. The generic way to get around this is to wrap/prefix all commands with `setarch linux32`, which resolves the issue in 100% of the cases I've come across.

But, that's not really relevant in this case, since you can just point to the existing cross-compile steps instead.

mtaunton, 2023-07-27:
@pelwell Thanks for taking the time to respond to this. 

> A 64-bit kernel build on a 32-bit OS is effectively a cross-compile, so a note saying to jump to "Cross-Compiling the Kernel" in that situation should be sufficient.

> My suggestion above remains unaltered.

But again, that is not the case I have reported. I wanted to build a 32-bit kernel on a (nominally) 32-bit OS. Any user might reasonably suppose (as I did) that this could simply be done by "Building the Kernel Locally". However, as I have reported, following the instructions that seem to apply for a Pi 400 / 4B / CM4 under "default 32-bit build configuration" actually leads to failure. The user has to know that in the current release of "32-bit" Raspberry Pi OS, the default running kernel is actually 64-bit, not 32-bit. Yet none of the instructions I followed mention this fact.

aallan, 2023-07-27:
> But again, that is not the case I have reported. I wanted to build a 32-bit kernel on a (nominally) 32-bit OS. Any user might reasonably suppose (as I did) that this could simply be done by "Building the Kernel Locally". 

Yup, and as @pelwell says, that's a docs problem we'll fix by telling folks that this isn't the case.

@pelwell Can you point me at exactly what point in the build flow we should point folks to the cross-compile instructions? Right at the start presumably?

Something along the lines of this…

````
NOTE: If you are attempting to compile a 32-bit kernel on the 32-bit distribution of Raspberry Pi OS you should be aware that while the userland in this distribution is 32-bit, the kernel is a 64-bit kernel. You should therefore follow the instructions for cross-compiling the kernel.
````

…work? Or would you like more?

pelwell, 2023-07-27:
OK, I'll amend my advice slightly:

> Following the cross-compilation instructions for the 64-bit _and 32-bit_ kernel removes the need for guesswork.

However, on the case of building 32-bit kernel on top of a 32-bit OS with a 64-bit kernel, the only necessary change is ARCH=arm

I think you should consider carefully your choice of kernel. The 64-bit kernel just runs better on the BCM2711.

aallan, 2023-07-27:
I've added an associated PR #3207 to resolve this issue.

mtaunton, 2023-07-27:
@pelwell, @aallan: thank you both.

Re building a 32-bit vs 64-bit kernel: I don't need the ultimate in performance for my application. But mainly I was worried that the code I am intending to develop from (which is someone else's work, and now fairly old) might not be 64-bit clean. Rather than (potentially) having to spend effort fixing such problems, it seemed simpler to keep everything 32-bit to begin with. However I'll certainly bear your comment in mind.

aallan, 2023-07-27:
...and documentation is updated with a hint and pointer to the cross-compiler instructions. Will go to the live site next time I push the `develop` branch to production.

PR #3027 (2023-07-27): Adding NOTE about building 32-bit kernel
If you are attempting to compile a 32-bit kernel on the 32-bit distribution of Raspberry Pi OS you should be aware that while the userland in this distribution is 32-bit, the kernel is a 64-bit kernel. You should therefore follow the instructions for cross-compiling the kernel.

Closes #3019 when merged.
closed 2023-07-27 by aallan