raspberrypi/documentation #3024: Pico hangs when debugging using Open-OCD from Homebrew

MrWFoulkes, 2023-07-26:
I followed the tutorial to set up a picoprobe on macos, and tested step-by-step debugging of the blink.c example. The tutorial gives the option of installing open ocd via homebrew or building from source. I first tried homebrew.

In gdb, after running `monitor reset init`, I'm unable to resume execution with `continue`. There's no error message; the pico seems to hang.

The solution was to uninstall open-ocd from homebrew and build from source.

aallan, 2023-07-26:
Not sure which tutorial you were following. If you see Appendix A of [Getting started with Raspberry Pi Pico](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf), where I talk about this, right at the start I say "For picoprobe to work, you need to build OpenOCD from the Raspberry Pi GitHub repository."

closed 2023-07-26 by aallan

MrWFoulkes, 2023-07-26:
@aallan This tutorial:
https://www.raspberrypi.com/documentation/microcontrollers/debug-probe.html#installing-openocd

aallan, 2023-07-26:
Hmm, okay thanks for that. I've fixed that up, in 807d506.