raspberrypi/documentation #3029: Create top level "legacy" config.txt box

aallan, 2023-07-27:
Move all the legacy `config.txt` options from the current `config.txt` page into a new top-level box to go on the Computers tab.

Need a box illustration from @jackbenwillis.

aallan, 2023-07-27:
> Hey @aallan what's the name of this new section?

So the current section is this

[image]

So, err, I'm going to go with "Legacy config.txt" ? So something that looks like this icon, but suggests... older?

@lurch was going to make an Asana task for you, but I guess if you saw this, it is not needful. :)

jackbenwillis, 2023-07-27:
[image]
This?

jackbenwillis, 2023-07-27:
Please don't use this...

jackbenwillis, 2023-08-01:
How about faded config.txt file?

aallan, 2023-08-01:
Works! @lurch when you spin up the new top box can you please use the icon above! :)

lurch, 2023-08-01:
Thanks @jackbenwillis !

closed 2023-08-01 by lurch

lurch, 2023-08-01:
[image]

commit 620bfd8f (2023-08-01): Skeleton commit for the new legacy config.txt options page.
Fixes #3029
commit 2054156d (2023-08-17): Skeleton commit for the new legacy config.txt options page.
Fixes #3029
commit 2cb0cddd (2023-09-06): Skeleton commit for the new legacy config.txt options page.
Fixes #3029
commit 21f42ab6 (2023-09-07): Skeleton commit for the new legacy config.txt options page.
Fixes #3029