raspberrypi/documentation #3034: Expand RP2040 design file section

aallan, 2023-08-01:
The [current section](https://www.raspberrypi.com/documentation/microcontrollers/rp2040.html#design-files) leaves a lot to be desired. Needs to be expanded. Should include pointers to the upcoming bookazine around the Hackspace [JLCPCB-friendly version](https://raspberrypi.social/@concretedog@mastodon.social/110814190940537756) of the Minimal Viable Board.

[image]

Reference to the bookazine should look like [this section](https://www.raspberrypi.com/documentation/computers/os.html#going-further) which reminds me, that needs templating and adding as a toolchain option like the White Papers (see #2973 for that).

aallan, 2023-11-26:
Still waiting for upstream and the Hackspace lot.

nllho, 2024-11-26:
Closing this as Won't Do: moving forward, we're going to avoid book callouts in this format.

closed 2024-11-26 by nllho

lurch, 2024-11-26:
Just for the benefit of anybody stumbling across this issue in future... the mentioned "bookazine" actually turned into a full book:
https://www.raspberrypi.com/news/new-book-release-design-an-rp2040-board-with-kicad/