raspberrypi/documentation #3037: Ruby 2.7 still needed?

aallan, 2023-08-01:
@nelliemckesson I can't remember, is the requirement for Ruby 2.7 to install the documentation toolchain still needed?

Didn't we get rid of this (or was that the *other* toolchain?). Came up during @jehlers42's onboarding.

lurch, 2023-08-01:
ping also @mudge as he setup the Github Actions stuff (and also dependabot?).

Yes, the _other_ toolchain has much better dependency-install steps (it uses locally-automatically-installed Python modules and Ruby gems, rather than relying on system-installed versions), but I didn't get a chance to port that nicer approach over to **this** toolchain yet.

aallan, 2023-08-01:
Yeah, and you probably won't get a chance any time soon. But I actually don't know whether Ruby 2.7 was ever a requirement for this toolchain, wasn't that an asciidoctor-pdf problem with the PDF toolchain? Urgh.

nelliemckesson, 2023-08-10:
OKAY @aallan confirming that ruby 2.7 is not needed

aallan, 2023-08-10:
> OKAY @aallan confirming that ruby 2.7 is not needed

OMG!

aallan, 2023-08-10:
Okay this is a reminder for me (or anyone that isn't sitting in a hotel bar trying to get through a week's worth of email) to update the [README.md](https://github.com/raspberrypi/documentation#readme) file so that it doesn't suggest that Ruby 2.7 is a requirement.

closed 2023-09-04 by aallan