raspberrypi/documentation #3044: Same-page anchors don't get checked

lurch, 2023-08-09:
Something I noticed while working on #2942 

If there's a link like `xref:legacy_config_txt.adoc#upstream_kernel[upstream_kernel]` then the toolchain will check that the `upstream_kernel` anchor appears _somewhere_ on the `legacy_config_txt.adoc` page (even if that page itself is made up of multiple included files).
However if there's a link like `<<overlay_prefix,overlay_prefix>>` (which creates a link to the `overlay_prefix` anchor on the same page), there's no check that the `overlay_prefix` anchor actually appears on the page.

I (luckily) spotted this because after the config.txt documentation-reorganisation, `documentation/asciidoc/computers/config_txt/boot.adoc` had a `<<upstream_kernel,upstream_kernel>>` link, which linked to a `upstream_kernel` anchor on the same page which no longer existed, the link had to be updated to `xref:legacy_config_txt.adoc#upstream_kernel[upstream_kernel]` instead.

It would be nice if the toolchain could check for `<<...>>` links which link to an anchor that doesn't exist.

aallan, 2023-09-04:
@nelliemckesson I think this is a you thing?

lurch, 2023-09-05:
> If there's a link like `xref:legacy_config_txt.adoc#upstream_kernel[upstream_kernel]` then the toolchain will check that the `upstream_kernel` anchor appears somewhere on the `legacy_config_txt.adoc` page (even if that page itself is made up of multiple included files).

@nelliemckesson For reference, see #2345

lurch, 2023-09-07:
Whilst adding additional link-checking functionality, it might also be worth checking that we don't have any `https://www.raspberrypi.com/documentation/blah` absolute-URLs in use (in the `.adoc` files), since they really ought to be relative-xrefs instead.

closed 2023-09-08 by aallan

lurch, 2023-09-11:
#3104 didn't solve the absolute-URL thing that I mentioned above, so I created #3106 for that.