raspberrypi/documentation #3061: Update all references to Raspberry Pi Press Books

aallan, 2023-08-22:
So with the integration of new toolchain functionality, see:

* #3035
* #3040

for details of that. There are three or four (or more?) instances where I've manually referenced Raspberry Pi Press books. These need ripping out and the new macro functionality used.

jehlers42, 2023-08-23:
Should I also update references to datasheets like these or just books?

https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf
https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-c-sdk.pdf

aallan, 2023-08-23:
Those aren't "books" they're part of the documentation. So won't work with the macro.

aallan, 2023-08-23:
It's stuff like https://www.raspberrypi.com/documentation/microcontrollers/micropython.html#where-can-i-find-documentation where we reference a Pi Press book that you're looking for.

aallan, 2023-08-23:
Or https://www.raspberrypi.com/documentation/computers/os.html#going-further

jehlers42, 2023-08-23:
Perfect, already captured those so expect a PR soon.

aallan, 2023-08-23:
#3070 closes this issue.