raspberrypi/documentation #3062: Integrate White Papers into documentation

aallan, 2023-08-22:
Some white papers have already been referenced in the documentation already, see [#2973](https://github.com/raspberrypi/documentation/issues/2973#issuecomment-1579289428) for how to do that. 

But there are still white papers that could be usefully inserted into the documentation and we haven't done that yet. Full list of published white papers at https://pip.raspberrypi.com/categories/685-whitepapers-app-notes.

lurch, 2023-08-22:
IIRC some whitepapers are only accessible if you've signed an NDA, and we probably shouldn't link to those from the documentation?

aallan, 2023-08-22:
No, but that link leads to a page of public White Papers.

jehlers42, 2023-08-25:
PR #3087 open for this. Discussion and opinions needed.