raspberrypi/documentation #3071: Integrate Raspberry Pi Press titles into documentation

aallan, 2023-08-23:
There are out-of-print titles at https://github.com/raspberrypipress/released-pdfs and still-in-print, titles at https://store.rpipress.cc/collections/books. 

Links to some of these exist in the documentation, and #3035 and #3040 added new functionality to easily link to them. 

In a similar fashion to #3062 where we're going to go looking for places to add in white papers, we should look for places that need links to any other Pi Press book we're not currently linking to.

nllho, 2024-11-26:
Closing this as Won't Do: moving forward, we're going to avoid doing book callouts in this format.

lurch, 2024-11-26:
> Closing this as Won't Do:

There's a `wontfix` label you can use for that :slightly_smiling_face: