raspberrypi/documentation #3075: Update Linux kernel version references

lurch, 2023-08-23:
There's multiple references (and URLs) to Linux kernel 5.4 scattered throughout the documentation (there's even a reference on https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#troubleshooting-spi-hardware to the Linux kernel 3.10!!).
These should all be updated to Linux kernel 6.1, which is used by both Bullseye and Bookworm.

jehlers42, 2023-08-23:
PR #3076

lurch, 2023-08-24:
I've seen the same on the pico-sdk repo - GitHub only auto-closes the issue linked to a PR when the change makes it all the way through to the `master` branch. And this PR only modified the `bookworm` branch.

aallan, 2023-08-24:
But I set the branch? 

[image]

so it should have done. I think? But I guess not.