raspberrypi/documentation #3080: Addition of a Power Supply page to the accessory documentation

aallan, 2023-08-24:
I was thinking we need a definitive power supplies page as part of the documentation that explains some of the issues around "chargers" vs "power supplies" which are fundamentally different beasts because they're trying to do different things. We have enough official power supplies now that things are more complicated than "buy the official power supply". So we need a page to point people towards

Can lift (move) existing discussion of powering various models from https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#power-supply to the new page. 

My standard blurb, which I used in the Getting Started section, on power supplies looks like this,

> We recommend the official Raspberry Pi Power Supply, which has been specifically designed to consistently provide +5.1V despite rapid fluctuations in current draw. Those fluctuations in demand are something that happens a lot when you’re using peripherals with the Raspberry Pi and something that other supplies—designed to provide consistent current for charging cellphones—usually don’t cope with all that well. It also has an attached micro USB cable, which means that you don’t accidentally use a poor-quality cable—something that can be an issue.

jehlers42, 2023-09-07:
Started work on this and #3000

aallan, 2023-11-26:
We now have the new 27W power supply for Pi 5. So this is more necessary than it was before!