raspberrypi/documentation #3081: request: BCM2711 technical specs datasheet

RaulTrombin, 2023-08-24:
Can't find technical specs such the max temperature, max temperature for the Ethernet controller and so on.

I found this report for a previous Raspberry PI:
https://copperhilltech.com/content/The%20Operating%20Temperature%20For%20A%20Raspberry%20Pi%20%E2%80%93%20Technologist%20Tips.pdf

lurch, 2023-08-24:
You may find https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#frequency-management-and-thermal-control useful too.

RaulTrombin, 2023-08-24:
Thanks for reply @aallan, but there I only have acess to this:
[datasheet](https://datasheets.raspberrypi.com/bcm2711/bcm2711-peripherals.pdf?_gl=1*1mzvykf*_ga*MTc1MTA5NzEyNS4xNjkyODg4NzUw*_ga_22FD70LWDS*MTY5Mjg4ODc1My4xLjEuMTY5Mjg5MTkxNi4wLjAuMA..)

Also did some searching on forum, and found this:
https://forums.raspberrypi.com/viewtopic.php?t=294815re

But not technical specs related datasheet.

aallan, 2023-08-24:
That is all the information we are able to release on the BCM2711.

RaulTrombin, 2023-08-24:
> You may find https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#frequency-management-and-thermal-control useful too.

@lurch thank you