raspberrypi/documentation #3089: /boot is moving to /boot/firmware

lurch, 2023-08-30:
So all references to `/boot/config.txt` and `/boot/cmdline.txt` will need updating to `/boot/firmware/config.txt`, `/boot/firmware/cmdline.txt`, etc. etc.

aallan, 2023-09-04:
@nathan-contino For `bookworm` labelled issues you'll need to check out the `bookworm` branch and work on there. Both @lurch and @jehlers42 are also working on the branch for push/pull as you go rather than waiting to do big commits otherwise you're going to run into merge conflicts.

fti-sfuke, 2023-09-04:
Hi @lurch @aallan 
Thank you for the update.

I've downloaded the 64-bit RPiOS 'Bookworm' from here: https://raspi.debian.net/tested-images/
Below are the environment details:
```
root@rpi4-20230612:/boot# cat /etc/os-release 
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```
```
root@rpi4-20230612:/boot# uname -a
Linux rpi4-20230612 6.1.0-9-arm64 #1 SMP Debian 6.1.27-1 (2023-05-08) aarch64 GNU/Linux
root@rpi4-20230612:/boot# 
```
I've noticed that there's no configuration-related content in the /boot/firmware/config.txt file. It only contains the following details:
```
root@rpi4-20230612:/boot/firmware# cat config.txt
# Do not modify this file!
#
# It is automatically generated upon install or update of either the
# firmware or the Linux kernel.
#
# If you need to set boot-time parameters, do so via the
# /etc/default/raspi-firmware, /etc/default/raspi-firmware-custom or
# /etc/default/raspi-extra-cmdline files.

# Switch the CPU from ARMv7 into ARMv8 (aarch64) mode
arm_64bit=1

enable_uart=1
upstream_kernel=1

kernel=vmlinuz-6.1.0-9-arm64
# For details on the initramfs directive, see
# https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=10532
initramfs initrd.img-6.1.0-9-arm64
```
```
root@rpi4-20230612:/boot/firmware# ls
bcm2711-rpi-400.dtb	  bcm2837-rpi-cm3-io3.dtb   fixup4.dat	  fixup_x.dat			   start4.elf	 sysconf.txt
bcm2711-rpi-4-b.dtb	  bcm2837-rpi-zero-2-w.dtb  fixup4db.dat  initrd.img-6.1.0-9-arm64	   start4x.elf	 user-data
bcm2711-rpi-cm4-io.dtb	  bootcode.bin		    fixup4x.dat   raspi-config_20121028_all.deb    start_cd.elf  vmlinuz-6.1.0-9-arm64
bcm2837-rpi-3-a-plus.dtb  cmdline.txt		    fixup_cd.dat  raspi-config_20121028_all.deb.1  start_db.elf
bcm2837-rpi-3-b.dtb	  config.txt		    fixup.dat	  start4cd.elf			   start.elf
bcm2837-rpi-3-b-plus.dtb  fixup4cd.dat		    fixup_db.dat  start4db.elf			   start_x.elf
```
I'm interested in modifying the RP4 configuration via the config.txt file that is mention in below:
We've successfully configured all of these operations on the Buster Raspi 32-bit OS.
```
# 3 Boot Options
#   B2 Wait for Network at Boot
#   B3 Splash Screen
#   P1 Camera
#   A6 Screen Blanking
#   A8 GL Driver
#   AA Pi 4 Video Output

raspi-config nonint do_boot_wait        1           # Disabled
raspi-config nonint do_boot_splash      1 '||' true # Disabled
raspi-config nonint do_wifi_country     $RASPI_WIFI_COUNTRY
raspi-config nonint do_camera           0       # Enabled
raspi-config nonint do_spi              0       # Enabled
raspi-config nonint do_i2c              0       # Enabled
raspi-config nonint do_serial           1       # Login & IF Disabled
raspi-config nonint do_onewire          0       # Enabled

raspi-config nonint do_memory_split     $RASPI_GPU_MEM
raspi-config nonint do_audio            $RASPI_AUDIO_OUTPUT
raspi-config nonint do_blanking         1       # Disabled
```
**Could you kindly advise where I can locate this information in Bookworm?**

aallan, 2023-09-04:
> I've downloaded the 64-bit RPiOS 'Bookworm' from here: https://raspi.debian.net/tested-images/

That isn't Raspberry Pi OS, that's a stock Debian image.

fti-sfuke, 2023-09-04:
@aallan Thanks for reply. 

Raspberry Pi OS Debian12 (  Bookworm ) is not yet available on the official Raspberry Pi website (https://www.raspberrypi.com/software/operating-systems/). 

And I have been unable to found a Raspberry Pi OS Debian 12 testing image. 
Could you kindly provide guidance on where I might obtain it? or should I wait for official Raspberry Pi OS release ?  

Is it not possible to achieve the above-mentioned operations on a stock Debian image?

aallan, 2023-09-04:
> Could you kindly provide guidance on where I might obtain it? or should I wait for official Raspberry Pi OS release ?

We are currently working internally towards a Bookworm release for Raspberry Pi. 

It is not yet available publically.

lurch, 2023-09-04:
> Is it not possible to achieve the above-mentioned operations on a stock Debian image?

That's a question for the maintainers of the image you're using :slightly_smiling_face: 

Raspberry Pi OS is _based_ on Debian, but with many changes; in a similiar way that Ubuntu is _based_ on Debian, but with many changes.

lurch, 2023-09-09:
Reopening, because of https://github.com/raspberrypi/documentation/pull/3105#issuecomment-1712482458

lurch, 2023-09-11:
Thanks @nathan-contino

closed 2023-09-11 by lurch

flatsiedatsie, 2023-11-17:
Out of curiosity: why was this done/needed? It's causing a lot of work to fix hardcoded references to /boot/ .

JamesH65, 2023-11-18:
I believe its an upstream Debian change.

ncopa, 2023-11-27:
> I believe its an upstream Debian change.

Do you have any references on why upstream debian did this change?

hhlocsync, 2024-03-03:
how to edit config.txt ?
 /boot/firmware/config.txt -> i edited but it not working

lurch, 2024-03-03:
@hhlocsync It sounds like your question is better suited to the forums https://forums.raspberrypi.com/