raspberrypi/documentation #3096: Need some meta-documentation

aallan, 2023-09-05:
After the Bookworm launch we probably need a `Writing.md` to go alongside the `README.md` that talks about how to use some of the macros we now have for the documentation site. So, things that probably need documenting:

* How to reference a white paper, see #2927
* How to reference a Raspberry Pi Press book, see #3035
* How to reference a tutorial, see #3060

Probably also need to document how to `xref:` sections between different pages, and how to reference sections inside the same page. Because of the way we use section titles as default anchors, and other issues it's not as obvious as it could be.

Can't think of anything else that needs documenting though, anyone?

lurch, 2023-09-05:
> Probably also need to document how to `xref:` sections between different pages, and how to reference sections inside the same page.

Also, because of the way that we build big pages out of multiple include files, it's not always obvious what is the "same page" or not :wink:

lurch, 2023-09-25:
Might also be worth adding some typographical style notes, i.e. when to use bold text, italic text, backticks, code-blocks, etc.

This would be so that we have consistent formatting for things like menu options (including submenus), program buttons, keypresses, program (executable) names, etc. etc.

aallan, 2023-09-25:
> Might also be worth adding some typographical style notes, i.e. when to use bold text, italic text, backticks, code-blocks, etc.
> 
> This would be so that we have consistent formatting for things like menu options (including submenus), program buttons, keypresses, program (executable) names, etc. etc.

Maybe. But I honestly don't want this to sprawl. That stuff can be fixed up fairly easily, where as cross references, or inserting content without using macros in some hack'ey sort of way is a pain in the backside.

lurch, 2023-10-11:
> This would be so that we have consistent formatting for things like menu options (including submenus), program buttons, keypresses, program (executable) names, command-sequences, example program output, etc. etc.

For example, #3152 uses both
```
Navigate to *Preferences* > *Raspberry Pi Configuration*.
```
and
```
Select `Display Options` > `Screen Blanking`.
```

and there's not (yet) any guidance about which of those is the "preferred" markup.

(but obviously this is still fairly low priority)

nathan-contino, 2023-10-11:
Note that I chose to use `monospace` text for `raspi-config` sort-of-GUI-via-CLI instructions and *bold* text for actual-GUI Raspberry Pi Configuration instructions. Our style guide should cover both!

aallan, 2023-10-11:
But in the first instance I want documentation on how to use the new macros, and some on how to `xref:` between documents, and how to link between sections in the same document. The rest can wait.

closed 2024-11-11 by nathan-contino