raspberrypi/documentation #3106: Check for absolute documentation URLs

lurch, 2023-09-11:
It might be worth checking that we don't have any https://www.raspberrypi.com/documentation/blah absolute-URLs in use (in the .adoc files), since they really ought to be relative-xrefs instead.
(obviously we'd need to check both `http:` and `https:`, and with and without the `www.`)

aallan, 2023-09-13:
One time grep through the existing documentation to check we haven't accidentally let something through in a PR.

lurch, 2023-09-13:
```
$ git grep "raspberrypi.com/documentation" "*.adoc"
documentation/asciidoc/accessories/audio/getting_started.adoc:In this example, we are using Raspberry Pi OS Lite. Our guides on https://www.raspberrypi.com/documentation/computers/getting-started.html#installing-the-operating-system[Getting started] cover this topic in great detail. Make sure that you update your operating system before proceeding and follow the instructions provided for Codec Zero configuration, including the commands to enable the on-board microphone and speaker output.
```

closed 2023-09-16 by aallan