raspberrypi/documentation #3108: How to tell which LED is ACT or Power

xgqfrms, 2023-09-12:
# How to tell which LED is ACT or Power

Red LED, long flash, no blink
Green LED, long flash, no blink

https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/computers/configuration/led_blink_warnings.adoc

lurch, 2023-09-12:
The red LED is power, the green LED is activity.

I can't tell what you're asking here, are you having a problem with your Raspberry Pi? If you are, the right place to ask for questions about that is at https://forums.raspberrypi.com/

closed 2023-09-12 by aallan

xgqfrms, 2023-09-12:
## It seems the Raspberry Pi has a bug

The green LED blink 6 times, and the red LED not blink (long flashes).

https://github.com/raspberrypi/rpi-eeprom/issues/484

https://www.youtube.com/watch?v=LlQJZUC2jCw