raspberrypi/documentation #3120: Why was the `avoid_warnings` config documentation removed?

dasl-, 2023-09-16:
Hi there,

I noticed that the documentation referencing the `avoid_warnings` `/boot/config.txt` option was removed last month: https://github.com/raspberrypi/documentation/commit/3865b6c842978758078be126f2e363af220a1ba4#diff-eefaba340bb7487511d89723ca7af494cf15889e339bdf9788f279b62d41c1c5

I'm curious why it was removed - does this config setting no longer work?

Thanks, and please let me know if I should ask this question in a different forum.

aallan, 2023-09-16:
See #2962.

closed 2023-09-16 by aallan

dasl-, 2023-09-16:
Thanks for the info. It seems like firmware warning _icons_ are no longer a thing. However, I think that CPU throttling may still occur if low voltage is present. I think that `avoid_warnings=2` prevents CPU throttling in the presence of low voltage. 

So I wonder if we still want to document the `avoid_warnings` setting for its use to prevent CPU throttling. Although it seems its use to prevent warning icons is no longer relevant.

Happy to be corrected if I am wrong about any of the above. Thanks.

lurch, 2023-09-17:
@dasl- See https://github.com/raspberrypi/documentation/blob/bookworm/documentation/asciidoc/computers/legacy_config_txt/misc.adoc

dasl-, 2023-09-17:
Ahh thank you. I seem to be unable to find reference to that at the documentation site for some reason: https://www.raspberrypi.com/documentation/computers/config_txt.html

But glad to see it still works!

lurch, 2023-09-18:
> I seem to be unable to find reference to that at the documentation site for some reason

Yes, the documentation at https://www.raspberrypi.com/documentation/ reflects the state of the [`master` branch](https://github.com/raspberrypi/documentation/tree/master), and there have recently been many changes on the [`bookworm` branch](https://github.com/raspberrypi/documentation/tree/bookworm) that haven't yet made their way onto `master` :slightly_smiling_face:

dasl-, 2023-09-18:
Makes sense. Thank you.

dasl-, 2025-10-28:
To anyone stumbling on this thread, here is where the documentation on `avoid_warnings` lives now: https://www.raspberrypi.com/documentation/computers/legacy_config_txt.html#avoid_warnings