raspberrypi/documentation #3128: De-emphasise `userconf.txt`

lurch, 2023-09-20:
Currently at https://www.raspberrypi.com/documentation/computers/configuration.html#setting-up-a-headless-raspberry-pi we have this boxout:
[image]

I _suspect_ that this was added before Raspberry Pi Imager gained the ability to add a new user account, as the above boxout makes it sound like creating `userconf.txt` is the preferred way of creating a new user. IMHO that boxout (and any other relevant sections on that page) should be re-ordered to **first** recommend configuring things using Raspberry Pi Imager, and then describe the manual methods afterwards.

jehlers42, 2023-09-20:
Suppose I could look at this as part of #2537?

lurch, 2023-10-12:
Fixed by #3166 :tada:

closed 2023-10-12 by lurch